ARMmbed/yotta #371: 'search' issue in yotta 0.6.0 under Windows 7

bogdanm, 2015-08-11:
```
yt search target
<snip>

additional results from private-registry:
<snip>

additional results from private-registry:
Traceback (most recent call last):
  File "C:\Python27\Scripts\yt-script.py", line 9, in <module>
    load_entry_point('yotta==0.6.0', 'console_scripts', 'yt')()
  File "c:\python27\lib\site-packages\yotta\main.py", line 218, in main
    status = args.command(args, following_args)
  File "c:\python27\lib\site-packages\yotta\search.py", line 69, in execCommand
    for result in registry_access.search(query=args.query, keywords=args.kw, registry=repo['url']):
  File "c:\python27\lib\site-packages\yotta\lib\registry_access.py", line 552, in search
    response.raise_for_status()
  File "c:\python27\lib\site-packages\requests\models.py", line 851, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found
```

autopulated, 2015-08-11:
Could you email me the search query that causes this?

bogdanm, 2015-08-11:
Sorry, that was a problem with my yotta configuration. Nothing wrong in yotta itself.

closed 2015-08-11 by bogdanm