ARMmbed/yotta #379: Yotta has dependencies on pre-release/development pip modules

sbutcher-arm, 2015-08-17:
When installing yotta on a clean Ubuntu 14.04 machine, yotta will not install cleanly without the user passing pip the --pre option because pyOCD has dependencies on a pre-release pyusb version.

This may be the intended behaviour, but in which case the installation documentation here: http://docs.yottabuild.org/#installing needs to be update it's documentation, or pyOCD needs to be less ambitious in it's dependencies.

0xc0170, 2015-08-17:
We should fix pyOCD requirements - using a specific pre-release version for now, as it seems pyUSB is not going to be stable that soon in the future. There's already PR for pyOCD, and also an issue tracking this pre-release issue.

Reference:
https://github.com/mbedmicro/pyOCD/pull/178

0xc0170, 2015-08-18:
@sbutcher-arm Can you please test? I released a new version of pyOCD (v0.5.0) which includes a fix related to this issue.

sbutcher-arm, 2015-08-18:
Retested on the same machine (Ubuntu 14.04) by uninstalling and reinstalling yotta with pip, and there is no longer an issue. Hence closing the issue.

closed 2015-08-18 by sbutcher-arm

autopulated, 2015-08-18:
:+1: Thanks!