ARMmbed/yotta #382: yt test fails but mbedgt --target=frdm-k64f-gcc does not

bridadan, 2015-08-18:
I'm testing an extremely bare bones project that only consists of one test that always passes.

When running the test with `mbedgt --target=frdm-k64f-gcc`, the test passes just fine.

When ran with `yt test`, the test fails with the following output:

```
$ yt test
warning: mbed_tet has invalid module.json:
warning:   homepage value u'' does not match u'^(https?://).*$'
info: generate for target: frdm-k64f-gcc 0.0.23 at /Users/arm/dev/mbed_tet/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/arm/dev/mbed_tet/build/frdm-k64f-gcc
ninja: no work to do.
info: test mbed-tet: mbed-tet-test-test1
MBED: Instrumentation: "/dev/tty.usbmodem11232" and disk: "/Volumes/MBED"
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Unknown property: meout;20}}
HOST: Property 'host_test_name' = 'default'
HOST: Property 'description' = 'Basic'
HOST: Property 'test_id' = 'MBED_A1'
HOST: Start test...
{{success}}
{{end}}
{{timeout;20}}
{{{{timeout;20}}
{{host_test_name;default}}
{{description;Basic}}
{{test_id;MBED_A1}}
{{start}}
{{success}}
{{end}}
error: test mbed-tet-test-test1 failed
info: tests complete: 0 passed, 1 failed
```

It looks like the serial buffer isn't getting properly flushed before reset? Not sure where the fix lies (or if it's just user error :) ) so I'm creating this issue for now.

autopulated, 2015-08-18:
Note that you'll need to define a test reporter script in your module (using, for example,  mbedgt --digest) to parse your tests output to determine success/failure.

See the yotta testing docs for details: http://docs.yottabuild.org/tutorial/testing.html

If you don't, then yotta will use the exit status of mbedhtrun, which will probably be nonzero due to timeout.

If you're already doing that, then please re-run with the open pull request that adds additional debug logging to yotta test, with "yotta -vvv test"

bridadan, 2015-08-18:
Ah, did not know about that! I'm curious, does it make sense to have mbedgt --digest to be the default test reporter script? Or would this cause confusion when building for x86 or other non-mbed platforms?

Also, that does seem to solve my problem, but only itermittently.

```
$ yt test
warning: mbed_tet has invalid module.json:
warning:   homepage value u'' does not match u'^(https?://).*$'
info: generate for target: frdm-k64f-gcc 0.0.23 at /Users/arm/dev/mbed_tet/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/arm/dev/mbed_tet/build/frdm-k64f-gcc
ninja: no work to do.
info: using filter "mbedgt --digest stdin -v -V" for tests in /Users/arm/dev/mbed_tet/build/frdm-k64f-gcc/test
info: test mbed-tet: mbed-tet-test-test1
MBED: Instrumentation: "/dev/tty.usbmodem11232" and disk: "/Volumes/MBED"
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Unknown property: $$$02400221D9340E1E24EAF3A6
mbed-host-test-runner: Stopped
error: test mbed-tet-test-test1 failed
info: tests complete: 0 passed, 1 failed
```

And this seems to be usually accompanied by a FAIL.TXT file on the mbed with the contents:

```
Flash algorithm write command FAILURE
```

I can actually reproduce this repeatedly. If I run `yt test` once, it works fine. If I run it again, I get the above error. Run it once more, works fine, etc. It flip flops.

At first I thought this was just a Mac issue (especially since I'm running in a VM), but I can't reproduce this problem by just running `mbedgt --target=frdm-k64f-gcc`

I will retry with the pull request you mentioned

bridadan, 2015-08-18:
Just retried with the #350 and that seems to do the trick! Consistently getting the following output:

```
$ yt test
warning: mbed_tet has invalid module.json:
warning:   homepage value u'' does not match u'^(https?://).*$'
info: generate for target: frdm-k64f-gcc 0.0.23 at /Users/arm/dev/mbed_tet/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/arm/dev/mbed_tet/build/frdm-k64f-gcc
ninja: no work to do.
info: using filter "mbedgt --digest stdin -v -V" for tests in /Users/arm/dev/mbed_tet/build/frdm-k64f-gcc/test
info: test mbed-tet: mbed-tet-test-test1
MBED: Instrumentation: "/dev/tty.usbmodem11232" and disk: "/Volumes/MBED"
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Property 'timeout' = '20'
HOST: Property 'host_test_name' = 'default'
HOST: Property 'description' = 'Basic'
HOST: Property 'test_id' = 'MBED_A1'
HOST: Start test...
{{success}}
{{end}}
{mbed-host-test-runner: Stopped
info: test mbed-tet-test-test1 passed
info: tests complete: 1 passed, 0 failed
```

It's worth noting that between `{mbed-host-test-runner: Stopped` being printed and yotta actually finishing, there is a good 5-6 second delay where nothing appears to happen.

Here's a snipped bit of the test with `-vvv` on:

```
info: using filter "mbedgt --digest stdin -v -V" for tests in /Users/arm/dev/mbed_tet/build/frdm-k64f-gcc/test
info: test mbed-tet: mbed-tet-test-test1
debug:root: run wrapped function...
debug:target: running test: [u'mbed_test_wrapper', u'--target', u'K64F', u'/Users/arm/dev/mbed_tet/build/frdm-k64f-gcc/test/mbed-tet-test-test1']
debug:target: using output filter command: [u'mbedgt', u'--digest', u'stdin', u'-v', u'-V']
MBED: Instrumentation: "/dev/tty.usbmodem11232" and disk: "/Volumes/MBED"
HOST: Copy image onto target...
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Property 'timeout' = '20'
HOST: Property 'host_test_name' = 'default'
HOST: Property 'description' = 'Basic'
HOST: Property 'test_id' = 'MBED_A1'
HOST: Start test...
{{success}}
{{end}}
{mbed-host-test-runner: Stopped
debug:target: test child trailing output: ""
debug:target: test mbed-tet-test-test1 passed
debug:root: wrapped function completed...
info: test mbed-tet-test-test1 passed
debug:root: inferred module mbed-alloc from path ym/mbed-alloc/test
debug:root: inferred module mbed-core from path ym/mbed-core/test
debug:root: inferred module minar from path ym/minar/test
debug:root: inferred module uvisor-lib from path ym/uvisor-lib/test
info: tests complete: 1 passed, 0 failed
```

autopulated, 2015-08-18:
Ok cool,

Yup we could make that the default once mbedgt is published, or we could set it as the default in initial module.json files (potentially pre-populating an empty test, too?)

With the debug output maxed, between which lines is the long delay?

bridadan, 2015-08-18:
The delay happens between `{mbed-host-test-runner: Stopped` and `debug:target: test child trailing output: ""`.

Maybe that unclosed curly brace is indicative of an issue somewhere?

autopulated, 2015-08-19:
OK, I've added even more debug logging in the updated PR.

I've also added a comprehensive set of tests for testReporter scripts, which exercises the (many!) possible combinations of test process and test script exiting before/after each other, with/without output etc. and check that tests don't take longer to run than expected. They all pass on OS X, I'm planning to get them (and for that matter the other tests that build things) running natively on windows soon too.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-895

bridadan, 2016-01-25:
I'm closing this issue since this exact behavior seems to be fixed in current versions of yotta and mbedgt

closed 2016-01-25 by bridadan