ARMmbed/yotta #385: 'yt lics' fails on a module without installed dependencies

bogdanm, 2015-08-24:
In a freshly cloned module that doesn't have dependencies installed yet, `yt lics` fails with this error message:

```Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 218, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/licenses.py", line 44, in execCommand
    for lic in dep.licenses():
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 586, in licenses
    return [x['type'] for x in self.description['licenses']]
KeyError: 'licenses'```

autopulated, 2015-09-07:
Thanks for reporting, this case should probably report that dependencies are missing so the result is incomplete, and recommend that the user runs `yotta install` first.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-894

PR #635 (2015-12-18): Fix crash in "yotta licenses" subcommand
 * display a nice error message instead of a backtrace if a module is missing
 * fixes #385
closed 2015-12-18 by autopulated