ARMmbed/yotta #387: Missing libffi dependency when installing yotta on Mac OS

sam-taylor-arm, 2015-08-26:
Just installed yotta on Mac OS 10.10.5 and hit two issues:
 * Missing cython dependency (already reported in mbedmicro/pyOCD#163)
 * Missing libffi dependency

The former is trivially resolved running `pip install cython`.

The later is resolved running `brew install libffi`. 

Terminal log follows for reference:

```
e107250-mac:~ samtay01$ pip install yotta
You are using pip version 7.1.0, however version 7.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Collecting yotta
Requirement already satisfied (use --upgrade to upgrade): semantic-version<3,>=2.3.1 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): Jinja2<3,>=2.7.0 in /usr/local/lib/python2.7/site-packages (from yotta)
Collecting argcomplete<1.0,>=0.8.0 (from yotta)
  Using cached argcomplete-0.9.0-py2.py3-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): requests<3,>=2.4.3 in /usr/local/lib/python2.7/site-packages (from yotta)
Collecting jsonschema<3.0,>=2.4.0 (from yotta)
  Using cached jsonschema-2.5.1-py2.py3-none-any.whl
Collecting cryptography>=0.8 (from yotta)
  Using cached cryptography-1.0.tar.gz
Requirement already satisfied (use --upgrade to upgrade): colorama<0.4,>=0.3 in /usr/local/lib/python2.7/site-packages (from yotta)
Requirement already satisfied (use --upgrade to upgrade): PyGithub<2,>=1.25 in /usr/local/lib/python2.7/site-packages (from yotta)
Collecting PyJWT<2.0,>=1.0 (from yotta)
  Using cached PyJWT-1.4.0-py2.py3-none-any.whl
Collecting mbed-test-wrapper<0.1.0,>=0.0.2 (from yotta)
Collecting valinor<1.0,>=0.0.0 (from yotta)
Requirement already satisfied (use --upgrade to upgrade): hgapi<2,>=1.7 in /usr/local/lib/python2.7/site-packages (from yotta)
Collecting intelhex<3.0,>=2.0 (from yotta)
  Using cached intelhex-2.0-py2.py3-none-any.whl
Collecting pathlib<1.1,>=1.0.1 (from yotta)
Requirement already satisfied (use --upgrade to upgrade): MarkupSafe in /usr/local/lib/python2.7/site-packages (from Jinja2<3,>=2.7.0->yotta)
Collecting functools32 (from jsonschema<3.0,>=2.4.0->yotta)
Requirement already satisfied (use --upgrade to upgrade): idna>=2.0 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.8->yotta)
Requirement already satisfied (use --upgrade to upgrade): pyasn1>=0.1.8 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.8->yotta)
Requirement already satisfied (use --upgrade to upgrade): six>=1.4.1 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.8->yotta)
Requirement already satisfied (use --upgrade to upgrade): setuptools in /usr/local/lib/python2.7/site-packages (from cryptography>=0.8->yotta)
Requirement already satisfied (use --upgrade to upgrade): enum34 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.8->yotta)
Requirement already satisfied (use --upgrade to upgrade): ipaddress in /usr/local/lib/python2.7/site-packages (from cryptography>=0.8->yotta)
Collecting cffi>=1.1.0 (from cryptography>=0.8->yotta)
  Using cached cffi-1.2.1.tar.gz
Collecting pyyaml<4,>=3 (from valinor<1.0,>=0.0.0->yotta)
Collecting pyOCD<1.0,>=0.3 (from valinor<1.0,>=0.0.0->yotta)
Collecting project-generator<0.8.0,>=0.7.0 (from valinor<1.0,>=0.0.0->yotta)
Requirement already satisfied (use --upgrade to upgrade): pycparser in /usr/local/lib/python2.7/site-packages (from cffi>=1.1.0->cryptography>=0.8->yotta)
Collecting hidapi (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
Collecting argparse (from project-generator<0.8.0,>=0.7.0->valinor<1.0,>=0.0.0->yotta)
  Using cached argparse-1.3.0-py2.py3-none-any.whl
Collecting xmltodict (from project-generator<0.8.0,>=0.7.0->valinor<1.0,>=0.0.0->yotta)
Building wheels for collected packages: cryptography, cffi
  Running setup.py bdist_wheel for cryptography
  Complete output from command /usr/local/opt/python/bin/python2.7 -c "import setuptools;__file__='/private/var/folders/yv/gm0nzvqj4v176t_w2rp5yk1svdyh2q/T/pip-build-gA_GRX/cryptography/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /var/folders/yv/gm0nzvqj4v176t_w2rp5yk1svdyh2q/T/tmpIyakpapip-wheel-:
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
[... 131 lines cut ...]
             ^
    1 error generated.
    error: command 'clang' failed with exit status 1

    ----------------------------------------
Command "/usr/local/opt/python/bin/python2.7 -c "import setuptools, tokenize;__file__='/private/var/folders/yv/gm0nzvqj4v176t_w2rp5yk1svdyh2q/T/pip-build-gA_GRX/cffi/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/yv/gm0nzvqj4v176t_w2rp5yk1svdyh2q/T/pip-dhvPAn-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/yv/gm0nzvqj4v176t_w2rp5yk1svdyh2q/T/pip-build-gA_GRX/cffi
e107250-mac:~ samtay01$ brew install libffi
==> Downloading https://homebrew.bintray.com/bottles/libffi-3.0.13.yosemite.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libffi-3.0.13.yosemite.bottle.tar.gz
==> Caveats
This formula is keg-only, which means it was not symlinked into /usr/local.

Some formulae require a newer version of libffi.

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/libffi/lib

==> Summary
🍺  /usr/local/Cellar/libffi/3.0.13: 14 files, 412K

```

0xc0170, 2015-09-01:
Related to pyOCD issue mentioned above - v0.5.1 which was released today should fix this

autopulated, 2015-09-07:
Closing this as it was a pyOCD issue, now fixed. Thanks for the detailed error report!

closed 2015-09-07 by autopulated