ARMmbed/yotta #388: Yotta CLI tool needs to be able to display announcements from the registry

hugovincent, 2015-08-28:
The yotta command line tool needs the ability to display announcements (e.g. banner/MOTD) received from the yotta registry. This will be necessary to provide a viable user experience for upcoming changes to the tool and to the registry. (Such messages will be infrequent).

Requirements:
* ability to test for and receive a message/banner/announcement text from the registry at least once per invocation that talks to the registry, and display it to the user.
* persistently store which announcements have been shown to users, so they don't get spammed with the same message over and over again.

autopulated, 2015-09-07:
ack.

We should figure out what touchpoints we expect users to have. I suspect everyone who  is using yotta will have several other regular touchpoints where they can learn things, too, and they should all mesh together.

thegecko, 2015-09-07:
Be aware these messages are likely to be 'hidden' in any IDE / editor