ARMmbed/yotta #389: Cannot start debugging in Windows

SeppoTakalo, 2015-08-28:
~~~~
$ yt debug mbed-client-example-6lowpan
←[22m←[32minfo←[39m: ←[22m←[39mfound mbed-client-example-6lowpan at source\mbed-
client-example-6lowpan←[0m
Traceback (most recent call last):
  File "c:\Python27\Scripts\valinor-script.py", line 9, in <module>
    load_entry_point('valinor==0.0.8', 'console_scripts', 'valinor')()
  File "C:\Python27\lib\site-packages\valinor\main.py", line 78, in main
    ide_tool = ide_detection.select(available_ides, args.target, project_setting
s)
  File "C:\Python27\lib\site-packages\valinor\ide_detection.py", line 126, in se
lect
    if targets.Targets(project_settings.get_env_settings('definitions')).is_supp
orted(target, ide):
  File "C:\Python27\lib\site-packages\project_generator\targets.py", line 37, in
 __init__
    self.targets = [ splitext(f)[0] for f in listdir(target_dir) if isfile(join(
target_dir,f)) ]
WindowsError: [Error 3] The system cannot find the path specified: 'c:/Users/sep
tak01\\.pg\\definitions\\target/*.*'
←[0m←[1m←[31merror←[39m: ←[22m←[31mdebug process exited with status 1←[0m
←[0m
~~~~

0xc0170, 2015-09-03:
@SeppoTakalo Is this still relevant? I recall reading somewhere you fixed this by doing update.

SeppoTakalo, 2015-09-03:
Yes, this is still relevant.

I managed to circumwent the problem by populating the content of `~/.pg` directory from my Linux machine.
So it was not about updating the tools. The tool responsible for populating the .pg directory seems to work on Linux but not on Windows.

0xc0170, 2015-09-03:
This might be a bug in project_generator (https://github.com/project-generator/project_generator). Please provide version number (project_generator --version).

SeppoTakalo, 2015-09-03:
~~~
C:\Users\septak01>project_generator --version
0.7.1
~~~

-- IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

ARM Limited, Registered office 110 Fulbourn Road, Cambridge CB1 9NJ, Registered in England & Wales, Company No: 2557590
ARM Holdings plc, Registered office 110 Fulbourn Road, Cambridge CB1 9NJ, Registered in England & Wales, Company No: 2548782

0xc0170, 2015-09-03:
Thanks ! That helped me to find a bug. It's in valinor, I'll send a fix shortly, and reference this one.

autopulated, 2015-09-07:
Closing as this is an issue in valinor.

closed 2015-09-07 by autopulated