ARMmbed/yotta #391: SSL: CERTIFICATE_VERIFY_FAILED

betzw, 2015-09-03:
Getting security error when trying just to build the very first tutorial example 'simplelog'. When typing `yotta build` the output result is:

```
warning: simplelog has invalid module.json:
warning:   homepage value u'' does not match u'^(https?://).*$'
info: get versions for x86-linux-native
Traceback (most recent call last):
  File "/usr/local/bin/yotta", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 229, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 41, in execCommand
    status = installAndBuild(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 64, in installAndBuild
    target, errors = c.satisfyTarget(args.target)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 574, in satisfyTarget
    update_installed = update_installed
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/target.py", line 97, in getDerivedTarget
    type = 'target'
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 354, in satisfyVersion
    return satisfyVersionByInstalling(name, version_required, working_directory, type=type)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 296, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 110, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 392, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 190, in _listVersions
    response = requests.get(url, headers=request_headers)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 69, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 50, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 465, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 573, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 431, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581)
```
To me it seems as it could be related to a generic Python problem regarding certificate verification and its implementation as described in [PEP476](https://www.python.org/dev/peps/pep-0476/) and documented [here](https://docs.python.org/3/library/http.client.html#http.client.HTTPSConnection).

A proper fix would be to use the systems default certificate store:
`context = ssl.create_default_context()`

What do you think?

autopulated, 2015-09-07:
The [requests](http://docs.python-requests.org/en/latest/user/advanced/) module that yotta uses does verify SSL certificates correctly. It seems likely that the certificate being served is failing verification for some other reason (such as using a root certificate that is missing from your system certificate store, or using a cipher suite that your module/python versions don't support).

Please could you check that your python version is at least 2.7.9, if not, does updating fix the issue?

(sorry for the delay in replying here, just got back from vacation ;)

betzw, 2015-09-07:
The problem for sure is related to a root certificate which is missing as we are behind a corporate firewall based on Zscaler, which typically manipulates certificates and requires you to have the Zscaler root certificate being installed under '/etc/ssl/certs' (at least in 'Ubuntu').
I am currently using python 2.7.8 which is the latest version currently supported by Ubuntu 14.10 on a 64bits machine and am sure that this version of Ubuntu does not use the certificates under '/etc/ssl/certs' but only those which come with the python installation itself. Upgrading to a newer version of python might resolve this issue, but as it is not supported by Ubuntu it would completely mix up my python installation and I would prefer to not having to do this.
Furthermore, the fix I proposed (i.e. using explicitly the system default certificate store) might be a solution even for python versions below 2.7.9 without compromising newer releases of python.

betzw, 2015-09-09:
Btw, in the meanwhile I have figured out that the problem is exactly the same on Windows using Python version 2.7.10!

autopulated, 2015-09-15:
Since I presume both computers you tested on are on your companies network it seems likely that the issue experienced in both cases is the same.

Unfortunately I'm not sure what yotta can do about this. The requests module uses a perfectly valid certificate store – it will also use [certifi](http://certifi.io/en/latest/) if it is present on the system, and the path to the certificate bundle can also be specified with the REQUESTS_CA_BUNDLE environment variable, but neither of these are things that yotta should be doing (as they could be highly user specific).

You could add your IT department's traffic introspection certificate to the certifi certificate store to work around the issue for now.

closed 2015-09-15 by autopulated