ARMmbed/yotta #394: build of helloyotta for frdm-k64f-gcc 0.0.24 fails

aspeteRakete, 2015-09-09:
Hi there,
in installed yotta 0.7.0 on osx. When trying to build helloyotta i get the following errors:
````
~/S/m/helloyotta (master)> yotta build
error: mbed-drivers does not meet specification ~0.6.0 required by minar
info: generate for target: frdm-k64f-gcc 0.0.24 at /Users/felix/Source/mbed/helloyotta/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/felix/Source/mbed/helloyotta/build/frdm-k64f-gcc
[129/129] Linking CXX executable source/helloyotta
FAILED: : && /usr/local/bin/arm-none-eabi-g++   -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -mcpu=cortex-m4 -mthumb -T"/Users/felix/Source/mbed/helloyotta/yotta_targets/frdm-k64f-gcc/CMake/../ld/K64FN1M0xxx12.ld" -Wl,-Map,source/helloyotta.map -Wl,--start-group source/CMakeFiles/helloyotta.dir/Users/felix/Source/mbed/helloyotta/source/hello.c.o  ym/simplelog/source/simplelog.a ym/mbed-drivers/source/mbed-drivers.a ym/mbed-hal/yotta_dummy_lib_mbed_hal/mbed-hal.a ym/minar/source/minar.a ym/core-util/source/core-util.a ym/mbed-hal-freescale/yotta_dummy_lib_mbed_hal_freescale/mbed-hal-freescale.a ym/minar-platform/yotta_dummy_lib_minar_platform/minar-platform.a ym/mbed-hal-ksdk-mcu/source/mbed-hal-ksdk-mcu.a ym/minar-platform-mbed/source/minar-platform-mbed.a ym/mbed-hal-k64f/source/mbed-hal-k64f.a ym/mbed-hal-frdm-k64f/source/mbed-hal-frdm-k64f.a ym/mbed-drivers/source/mbed-drivers.a ym/mbed-hal/yotta_dummy_lib_mbed_hal/mbed-hal.a ym/minar/source/minar.a ym/core-util/source/core-util.a ym/mbed-hal-freescale/yotta_dummy_lib_mbed_hal_freescale/mbed-hal-freescale.a ym/minar-platform/yotta_dummy_lib_minar_platform/minar-platform.a ym/mbed-hal-ksdk-mcu/source/mbed-hal-ksdk-mcu.a ym/minar-platform-mbed/source/minar-platform-mbed.a ym/mbed-hal-k64f/source/mbed-hal-k64f.a ym/mbed-hal-frdm-k64f/source/mbed-hal-frdm-k64f.a ym/compiler-polyfill/yotta_dummy_lib_compiler_polyfill/compiler-polyfill.a ym/ualloc/source/ualloc.a ym/dlmalloc/source/dlmalloc.a -Wl,-wrap,_malloc_r -Wl,-wrap,_calloc_r -Wl,-wrap,_realloc_r -Wl,-wrap,_free_r -Wl,-wrap,_sbrk_r ym/cmsis-core/yotta_dummy_lib_cmsis_core/cmsis-core.a ym/cmsis-core-freescale/yotta_dummy_lib_cmsis_core_freescale/cmsis-core-freescale.a ym/cmsis-core-k64f/source/cmsis-core-k64f.a ym/cmsis-core/yotta_dummy_lib_cmsis_core/cmsis-core.a ym/cmsis-core-freescale/yotta_dummy_lib_cmsis_core_freescale/cmsis-core-freescale.a ym/cmsis-core-k64f/source/cmsis-core-k64f.a ym/uvisor-lib/source/uvisor-lib.a -lstdc++ -lsupc++ -lm -lc -lgcc -lstdc++ -lsupc++ -lm -lc -lgcc -Wl,--end-group  --specs=nano.specs -o source/helloyotta && cd /Users/felix/Source/mbed/helloyotta/build/frdm-k64f-gcc/source && /usr/local/bin/arm-none-eabi-objcopy -O binary helloyotta helloyotta.bin && cd /Users/felix/Source/mbed/helloyotta/build/frdm-k64f-gcc/source && /usr/local/bin/arm-none-eabi-objcopy -O binary helloyotta helloyotta.bin
ym/mbed-drivers/source/mbed-drivers.a(retarget.cpp.o): In function `main':
retarget.cpp:(.text.startup.main+0x0): multiple definition of `main'
source/CMakeFiles/helloyotta.dir/Users/felix/Source/mbed/helloyotta/source/hello.c.o:/Users/felix/Source/mbed/helloyotta/source/hello.c:17: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

The potential lines with useful information are:
````

error: mbed-drivers does not meet specification ~0.6.0 required by minar
````
and
````
In function `main':
retarget.cpp:(.text.startup.main+0x0): multiple definition of `main'
```

0xc0170, 2015-09-09:
This should be reported to the module which you were building (helloyotta). There are 2 main defined, as mbed drivers module defines main. I'll send fix to helloyotta shortly.

aspeteRakete, 2015-09-09:
Ok then let's make this about the minar error.
````
cat mbed-drivers/module.json
{
  "name": "mbed-drivers",
  "version": "0.7.0",
  "description": "mbed SDK",
  "keywords": [
    "mbed"
  ],
  "author": "Bogdan Marinescu <bogdan.marinescu@arm.com>",
  "repository": {
    "url": "git@github.com:ARMmbed/mbed-drivers.git",
    "type": "git"
  },
  "homepage": "https://github.com/ARMmbed/mbed-drivers",
  "licenses": [
    {
      "url": "https://spdx.org/licenses/Apache-2.0",
      "type": "Apache-2.0"
    }
  ],
  "extraIncludes": [
    "mbed"
  ],
  "dependencies": {
    "mbed-hal": "~0.6.0",
    "cmsis-core": "~0.2.0",
    "ualloc": "~0.0.0",
    "minar": "~0.6.0",
    "core-util": "~0.0.11",
    "compiler-polyfill": "~1.0.0"
  },
  "targetDependencies": {},
  "scripts": {
    "testReporter": [
      "mbedgt",
      "--digest",
      "stdin",
      "-v",
      "-V"
    ]
  }
}
````
> cat minar/module.json
{
  "name": "minar",
  "version": "0.6.6",
  "description": "The mbed OS scheduler.",
  "repository": {
    "type": "git",
    "url": "git@github.com:ARMmbed/minar.git"
  },
  "homepage": "https://github.com/ARMmbed/minar",
  "bugs": {
    "url": "http://github.com/ARMmbed/minar/issues",
    "email": "project@hostname.com"
  },
  "author": "James Crosby <James.Crosby@arm.com>",
  "licenses": [
    {
      "url": "https://spdx.org/licenses/Apache-2.0",
      "type": "Apache-2.0"
    }
  ],
  "dependencies": {
    "compiler-polyfill": "~1.0.3",
    "minar-platform": "~0.3.0",
#include "mbed/mbed.h"
    "mbed-drivers": "~0.6.0",
    "core-util": "~0.0.6"
  },
  "scripts": {
    "testReporter": [
      "mbedgt",
      "--digest",
      "stdin",
      "-v",
      "-V"
    ]
  }
}
`````

0xc0170, 2015-09-09:
Ah that minar error req is also about to be fixed. I sent PR today, to update dependency - https://github.com/ARMmbed/minar/pull/4. We are waiting for 2 more PR to be complete/merged to avoid this circular dependency (minar - mbed-drivers). Thanks for reporting

aspeteRakete, 2015-09-09:
Ok. Thanks for clearing that up. [example-mbedos-blinky](https://github.com/ARMmbed/example-mbedos-blinky) compiled fine even without the minar error.

closed 2015-09-09 by aspeteRakete