ARMmbed/yotta #403: build information as compile time defines

jupe, 2015-09-15:
there should be several defines which YOTTA generates build time:

* YOTTA_BUILD_DATE
* YOTTA_BUILD_TIME
* YOTTA_BUILD_SCM_HASH   (GIT/HG HASH)
* YOTTA_BUILD_UUID (generated by yotta)

Those are very important for example when verify that software under test is what we expect..

jupe, 2015-09-15:
Looks that this affects only to this file:
https://github.com/ARMmbed/yotta/blob/master/yotta/lib/cmakegen.py#L213

autopulated, 2015-09-15:
The challenge here is that we don't want to re build everything every time the config header changes, so this can't be put in the config header (or on the command line).

Instead I'm thinking of putting this in a separate yotta-generated header which is *not* automatically included, and which you'll have to explicitly include.

If this header is generated in the root of the build directory (like the config header), then there are two options for how to include it:

 1. yotta defines the path to the file as a -D definition, which you can then `#include YOTTA_BUILD_INFO_HEADER`, or similar. The challenge here is that CMake will not track the dependency on this file, so if you want the object including that header to be rebuilt on every build then you will have to add custom CMake to explicitly add that dependency.

2. we could require custom CMake to include the file at all (path constructed in cmake as `"${CMAKE_BINARY_DIR}/yotta_build_info.h"`), since you're probably going to want to do custom CMake anyway

just thinking out loud...

jupe, 2015-09-15:
it's good to thinking out loud! I see your point. This is not as simple as I originally was thinking but luckily you give already two solutions :+1:

PR #404 (2015-09-16): Working on resolving #403
Resolves #403
closed 2015-09-16 by autopulated