ARMmbed/yotta #407: collect yotta module sizes to get statistics of memory usage

jupe, 2015-09-16:
It would be very useful if yotta could collect module sizes after compiling (with configuration details). Then we could estimate easily how much dependencies takes memory and how much there is free space left on selected target for user software. It could be also used to testing point of view: we could easily follow the memory usage while users develop modules and that gives valuable information, for example optimization point of view. Of course this could be part of CI, but then it's not so generic feature..

hugovincent, 2015-09-16:
IMO, this is a task for the base CMake files in https://github.com/ARMmbed/target-mbed-gcc and https://github.com/ARMmbed/target-mbed-armcc, rather than of Yotta itself, although maybe yotta should support an optional post-process script (in the style of the `testReporter` [script](http://yottadocs.mbed.com/reference/module.html#scripts)) for presenting the results nicely.

jupe, 2015-09-16:
well, CMake do actual job to fetch binary sizes, but yotta should collect those and upload meta information to the cloud for example.. of course if there is nice reported integrated to yotta post-process script, it could be useful as well. but yes, at least this affect to cmake part..

autopulated, 2015-09-16:
yotta definitely needs to enable other tools to do those things, but I'm not clear on what's necessary to do that beyond the cmake post-build command support that already exists?

Collecting size statistics for different modules on different platforms is something that a continuous integration and test service would do, I think.

jupe, 2015-09-16:
Maybe that nice memory report could be integrate to yotta. I agree that CI and test service could do the job, but it could give valuable information if we could collect build statistics directly from yotta..

autopulated, 2015-09-17:
I'm closing this because I don't believe it's something that yotta should add any more support for above that which already exists.

closed 2015-09-17 by autopulated

hugovincent, 2015-09-17:
@jupe maybe file an issue on [target-mbed-gcc](https://github.com/ARMmbed/target-mbed-gcc) and/or [target-mbed-armcc](https://github.com/ARMmbed/target-mbed-armcc) instead? I think that's the right place to do it.

jupe, 2015-09-18:
I agree, target-mbed-gcc and armcc is the place where memory fetch functionality need to be done, but I think you little misunderstand the bigger idea behind my question. If we could collect these information to our yotta-service it could give much very valuable information automatically from all of yotta builds, not just from continuous integration. E.g. some user start develop application above the mbedOS. He select some useful yotta modules, e.g. some connectivity and service modules. Then he start thinking how much those modules takes memory, is it going to fit for his device.. With this statistics information, yotta could tell directly that those selected modules takes X-bytes memory with current configurations and you have still Y-bytes memory left for application... if Y -value is negative value, probably user select too many modules and he need to give up some of those because otherwise it is not going to fit to the device....

mlnx, 2015-10-08:
+1
As already explained by @jupe above, many people would benefit from this info

hugovincent, 2015-10-08:
As per comments above, displaying code size etc isn't a role for yotta itself. But it sounds like there is also a separate idea above from @jupe to collect analytics data and send back to the yotta registry so it can provide information like average size (across targets/configurations, or for a specific target). This might be a good idea, lets open a separate ticket to track it (since we probably won't be implementing it soon).

bogdanm, 2015-10-08:
Note that this kind of information can be confusing at times. At link time, all the things that you don't use from a particular module are stripped away. So it might be that a module appears to be large in terms of memory, but your application uses only a small subset of that module, so your initial information about size is misleading.

hugovincent, 2015-10-08:
Yes, that's right. Ideally, if we were to implement this, it would track which module each function and data section in the final linked binary came from, and only count that (rather that the size of the `.a`/etc file, much of which will be stripped away during link-time garbage collection).