ARMmbed/yotta #408: RuntimeError: OrderedDict mutated during iteration (Python3.5)

jserv, 2015-09-16:
While running `yotta` with [python3.5](https://packages.debian.org/sid/python3.5) on Debian GNU/Linux, `yotta` stops and complains "RuntimeError: OrderedDict mutated during iteration".

The way to reproduce:
```
git clone git@github.com:ARMmbed/mbed-drivers.git && cd mbed-drivers
yotta target frdm-k64f-gcc
yotta -v build
```
Traceback:
```
  File "/usr/local/bin/yotta", line 4, in <module>
    __import__('pkg_resources').run_script('yotta==0.7.0', 'yotta')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 735, in run_script
...
    return installDeps(args, c)
  File "/usr/local/lib/python3.5/dist-packages/yotta-0.7.0-py3.5.egg/yotta/install.py", line 105, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "/usr/local/lib/python3.5/dist-packages/yotta-0.7.0-py3.5.egg/yotta/lib/component.py", line 559, in satisfyDependenciesRecursive
    test = test
  File "/usr/local/lib/python3.5/dist-packages/yotta-0.7.0-py3.5.egg/yotta/lib/component.py", line 375, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python3.5/dist-packages/yotta-0.7.0-py3.5.egg/yotta/lib/component.py", line 375, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python3.5/dist-packages/yotta-0.7.0-py3.5.egg/yotta/lib/component.py", line 375, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python3.5/dist-packages/yotta-0.7.0-py3.5.egg/yotta/lib/component.py", line 366, in __getDependenciesRecursiveWithProvider
    for c in need_recursion:
RuntimeError: OrderedDict mutated during iteration
```

autopulated, 2015-09-16:
Thanks for reporting this – python 3 support is currently experimental, so it looks like this is something the current test suite doesn't cover.

closed 2015-09-17 by autopulated

guillochon, 2016-01-11:
Hi, so I encountered this issue with my own code, and I was performing some `del` operations on the `OrderedDict` while looping through it. My solution was to perform a `copy.deepcopy` of the `OrderedDict` and perform the `del` operations on that, and now I don't have the issue anymore. So perhaps something like that was going on here.

autopulated, 2016-01-12:
In this case there was a difference between a generator and an explicit list between python 2 and 3 (where in the python 3 case because the list being iterated over was really a generator that was a view onto the ordered dictionary, modifying the dictionary during iteration would cause the error): this is the line that was changed to fix it:

https://github.com/ARMmbed/yotta/pull/414/files#diff-699129f8cde34ce5a928fa71a8f1a68cL359

(Basically explicitly converting the generator into an actual list)

:)