ARMmbed/yotta #409: linker error when using module as a test dependency

rgrover, 2015-09-16:
I get linker errors on x86-linux-native when using cpputest as a test framework. If I make cpputest as a direct dependency, the tests get built properly, but not if I turn them into a targetDepency.

Here's the sort of linker error I get:

```info: generate for target: x86-linux-native 0.0.6 at /tmp/cpputest/yotta_targets/x86-linux-native
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/cpputest/build/x86-linux-native
[5/5] Linking CXX executable test/mbed-client-trace-test-firsttest
FAILED: : && /usr/bin/c++   -O2 -g -DNDEBUG   test/CMakeFiles/mbed-client-trace-test-firsttest.dir/tmp/cpputest/test/FirstTest/FirstTest.cpp.o test/CMakeFiles/mbed-client-trace-test-firsttest.dir/tmp/cpputest/test/FirstTest/main.cpp.o  -o test/mbed-client-trace-test-firsttest  -rdynamic ym/cpputest/source/libcpputest.a yotta_dummy_lib_mbed_client_trace/libmbed-client-trace.a && :
test/CMakeFiles/mbed-client-trace-test-firsttest.dir/tmp/cpputest/test/FirstTest/FirstTest.cpp.o:(.rodata._ZTV39TEST_FirstTestGroup_FirstTest_TestShell[_ZTV39TEST_FirstTestGroup_FirstTest_TestShell]+0x30): undefined reference to `UtestShell::isNull() const'
```

Here's the module.json which uses cpputest as a test-dependency.
```
{
  "name": "mbed-client-trace",
  "version": "0.0.1",
  "description": "Simple Trace library for mbed clients",
  "keywords": [
    "trace",
    "debug",
    "library",
    "client"
  ],
  "author": "Jussi Vatjus-Anttila",
  "repository": {
    "url": "https://github.com/ARMmbed/mbed-client-trace.git",
    "type": "git"
  },
  "homepage": "https://github.com/ARMmbed/mbed-client-trace",
  "licenses": [
    {
      "url": "https://spdx.org/licenses/Apache-2.0",
      "type": "Apache-2.0"
    }
  ],
  "dependencies": {
  },
  "testDependencies": {
    "cpputest": "*"
  }
}
```

The cpputest module on github/armmbed doesn't support the gcc platform. I'll send you a tar ball for what I've got.

rgrover, 2015-09-16:
the linker command line in the two cases differs, but there's something more to it than that. I could edit the linker command line in the failing case and get the link to succceed.

autopulated, 2015-09-17:
could you send me the examples that do/don't work?

closed 2015-09-21 by autopulated

rgrover, 2015-09-22:
works for me now. Thanks!

autopulated, 2015-09-22:
ahhh, ok, cool!

jupe, 2015-09-24:
is it really working ? it doesn't work for me :s

PR #418 (2015-09-21): ensure that include directories of test dependencies are absolute
Fix case where testDependencies with additional include directories specified would fail to compile.

Possibly fixes #409 

@rgrover please confirm if this resolves your issue. It resolves everything that I can reproduce, but shouldn't technically have changed anything related to linking.