ARMmbed/yotta #413: yotta install can't install mbed-cpputest module

cyliangtw, 2015-09-17:
I tried mbed example "mbed-example-asynch-spi" to verify installation of dependent module "mbed-cpputest" by yotta but encountered  "does not exist in the modules registry". In this sample, it only depends on "mbed-cpputest" & "mbed" module, how could I complete yotta_modules by 'yotta install' ?
Show the command & replied error as below:
C:\mbedOS-alpha3\examples\mbed-ex
ample-asynch-spi>yotta install mbed-cpputest
info: get versions for mbed-cpputest
error: mbed-cpputest does not exist in the modules registry. Check that the name
 is correct, and that it has been published.

C:\\mbedOS-alpha3\examples\mbed-ex
ample-asynch-spi>yotta install
info: get versions for mbed
info: get versions for mbed-cpputest
error: Failed to satisfy dependencies of C:\Users\cyliang.NUVOTON\Desktop\ARM_IO
T\ARM_mbed\mbedOS-alpha3\examples\mbed-example-asynch-spi:
error: mbed does not exist in the modules registry. Check that the name is corre
ct, and that it has been published.
error: mbed-cpputest does not exist in the modules registry. Check that the name
 is correct, and that it has been published.

0xc0170, 2015-09-17:
Are you using alpha3 release? That release used prefetched modules, therefore yotta install fails as those dependencies have been changed since alpha3, in most cases.

Please use https://github.com/ARMmbed/example-asynch-spi

cyliangtw, 2015-09-17:
Hi Martin,
        Yes, I am using alpha3 release.
After used your proposed example-asynch-spi. I could finish yotta install except getting the below error.
Do you have any idea to avoid this un-match issue ?

========================================================================
C:\ mbedOS-alpha3\examples\example-asynch-spi-master>yotta install mbed-drivers
info: dependency mbed-drivers is already present in module.json
error: mbed-drivers does not meet specification ~0.7.0 required by minar

BR.
cyliang

The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

0xc0170, 2015-09-17:
I'll update examples, thanks for reporting. There is a newer version which resolved this dependency error.

autopulated, 2015-09-17:
Closing this issue as it relates to example applications, not yotta. Please re-open if this is erroneous.

closed 2015-09-17 by autopulated