ARMmbed/yotta #419: yotta test mbed-test-stl failed

cyliangtw, 2015-09-21:
To run yotta test based on FRDM K64F target, it could pass mbed-test-hello but got error as "error: test mbed-test-stl failed". For clarification, I executed mbed-test-stl by manual drag  & drop the bin file instead of through yotta test, after reset, the sort equal result showed true & notified 'success' as below. What's wrong with yotta test or my environment ?
---------------------------------------------------
[OK] Fill vector with data
[OK] Transform float strings
0.002 0.92430 15.91320 791.77368 6208.2 25719.4952 426815.982588 6429271.046 424
68024.93 212006462.910
0.002000 0.924300 15.913200 791.773682 6208.200195 25719.496094 426815.968750 64
29271.000000 42468024.000000 212006464.000000
[OK] Sort integers
99999 32760 4231 4123 999 1 0 -1 -100 -999 -3214 -32768
99999 32760 4231 4123 999 1 0 -1 -100 -999 -3214 -32768
[OK] Sort integers
99999 -32768 32760 4231 4123 -3214 999 -999 -100 1 -1 0
99999 -32768 32760 4231 4123 -3214 999 -999 -100 1 -1 0
{{success}}
           {{end}}

autopulated, 2015-09-21:
Thanks for this report – please could you provide:

 * the log output when you run `yotta test`
 * the name and version of the OS that you use
 * the output of `pip list`

This should help us identify the problem.

cyliangtw, 2015-09-21:
* log output when run yotta test: 
```
c:\mbedOS-alpha3\libraries\mbed-sdk>yotta test mbed-test-stl
info: generate for target: frdm-k64f-armcc 0.0.11 at c:\mbedOS-alpha3\libraries\mbed-sdk\yotta_targets\frdm-k64f-armcc
mbedOS.cmake included
ARMCC-C.cmake included
mbedOS-GNU-C.cmake included
ARMCC-CXX.cmake included
mbedOS-GNU-CXX.cmake included
ARMCC-ASM.cmake included
-- Configuring done
-- Generating done
-- Build files have been written to: C:/mbedOS-alpha3/libraries/mbed-sdk/build/frdm-k64f-armcc
ninja: no work to do.
info: using filter "mbedgt --digest stdin -v -V" for tests in c:\mbedOS-alpha3\libraries\mbed-sdk\build\frdm-k64f-armcc\test
info: test mbed: mbed-test-stl
mbedgt: selecting test case observer...
        selected digest source: stdin
MBED: Instrumentation: "COM1" and disk: "D:"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Unknown property: [OK] Fill vector with data
HOST: Unknown property: [OK] Transform float strings
HOST: Unknown property: 0.002 0.92430 15.91320 791.77368 6208.2 25719.4952 42681
5.982588 6429271.046 42468024.93 212006462.910
HOST: Unknown property: 0.002000 0.924300 15.913200 791.773682 6208.200195 25719
.496094 426815.968750 6429271.000000 42468024.000000 212006464.000000
HOST: Unknown property: [OK] Sort integers
Hmbedgt: mbed-host-test-runner: stopped
Completed in 7.59 sec
error: test mbed-test-stl failed
info: tests complete: 0 passed, 1 failed
```
    * name and version of the OS: Win7
    * output of pip list:
```
c:\mbedOS-alpha3\libraries\mbed-sdk>pip list
argcomplete (0.9.0)
cffi (1.2.1)
colorama (0.3.3)
cryptography (1.0.1)
enum34 (1.0.4)
functools32 (3.2.3.post2)
hgapi (1.7.2)
idna (2.0)
intelhex (2.0)
ipaddress (1.0.14)
Jinja2 (2.8)
jsonschema (2.5.1)
junit-xml (1.6)
lockfile (0.10.2)
MarkupSafe (0.23)
mbed-greentea (0.1.3)
mbed-host-tests (0.1.15)
mbed-ls (0.1.19)
mbed-test-wrapper (0.0.2)
ntfsutils (0.1.3)
pathlib (1.0.1)
pip (7.0.1)
prettytable (0.7.2)
project-generator (0.7.2)
pyasn1 (0.1.8)
pycparser (2.14)
PyGithub (1.25.2)
PyJWT (1.4.0)
pyOCD (0.5.1)
pyserial (2.7)
pywinusb (0.3.6)
PyYAML (3.11)
requests (2.7.0)
semantic-version (2.4.2)
setuptools (18.3.1)
shutilwhich (1.1.0)
six (1.9.0)
valinor (0.0.9)
wget (2.2)
xmltodict (0.9.2)
yotta (0.7.0)
```

PrzemekWirkus, 2015-09-21:
Hello,
Prints from log:
```
...
HOST: Reset target...
HOST: Detecting test case properties...
HOST: Unknown property: [OK] Fill vector with data
HOST: Unknown property: [OK] Transform float strings
HOST: Unknown property: 0.002 0.92430 15.91320 791.77368 6208.2 25719.4952 42681
...
```
Suggest that some preamble test case prints are not present (see below).
Full output from binary when flashed manually (STL test case) should look like this:
```
{{timeout;10}}
{{host_test_name;default_auto}}
{{description;C++ STL}}
{{test_id;MBED_A3}}
{{start}}
[OK] Fill vector with data
[OK] Transform float strings
0.002 0.92430 15.91320 791.77368 6208.2 25719.4952 426815.982588 6429271.046 42468024.93 212006462.910

[OK] Sort integers
99999 32760 4231 4123 999 1 0 -1 -100 -999 -3214 -32768
99999 32760 4231 4123 999 1 0 -1 -100 -999 -3214 -32768
[OK] Sort integers
99999 -32768 32760 4231 4123 -3214 999 -999 -100 1 -1 0
99999 -32768 32760 4231 4123 -3214 999 -999 -100 1 -1 0
{{success}}
{{end}}
```
Missing OP's (preamble test case) prints:
```
{{timeout;10}}
{{host_test_name;default_auto}}
{{description;C++ STL}}
{{test_id;MBED_A3}}
{{start}}
...
```
that are used to "handshake" between test suite running on your host computer.
It may suggest you do not have latest mbed-drivers ?

Macros responsible for "test handshake" prints are here:
https://github.com/ARMmbed/mbed-drivers/blob/master/test/stl/main.cpp#L83
They should be compiled into binary we are flashing on the device.

@cyliangtw Could you confirm that preamble prints are missing?

cyliangtw, 2015-09-21:
Yes, preamble prints from timeout~ start are missing.
After added the below snippet into main() function, it's working. Thanks.

```
    MBED_HOSTTEST_TIMEOUT(10);
    MBED_HOSTTEST_SELECT(default_auto);
    MBED_HOSTTEST_DESCRIPTION(C++ STL);
    MBED_HOSTTEST_START("MBED_A3");	
```

closed 2015-09-21 by cyliangtw