ARMmbed/yotta #425: target.json schema should support dictionaries

bremoran, 2015-09-23:
Supporting dictionaries is necessary for some kinds of config to be meaningful, such as pin definitions.  This is a requirement for https://github.com/ARMmbed/mbed-hal/pull/23

autopulated, 2015-09-23:
looking at the schema, I'm not sure what it doesn't allow which is necessary? (It should allow nested "targetData" constructs)

See here: https://github.com/ARMmbed/yotta/blob/master/yotta/lib/schema/target.json#L297

bremoran, 2015-09-23:
Output when building https://github.com/ARMmbed/target-frdm-k64f-gcc/pull/9 with is:
```
warning: frdm-k64f-gcc has invalid target.json:
warning:   config.mbed value OrderedDict([(u'pins', OrderedDict([(u'LED_RED', u'PTB22'), (u'LED_GREEN', u'PTE26'), (u'LED_BLUE', u'PTB21'), (u'LED1', u'LED_RED'), (u'LED2', u'LED_GREEN'), (u'LED3', u'LED_BLUE'), (u'LED4', u'LED_RED'), (u'SW2', u'PTC6'), (u'SW3', u'PTA4'), (u'USBTX', u'PTB17'), (u'USBRX', u'PTB16'), (u'D0', u'PTC16'), (u'D1', u'PTC17'), (u'D2', u'PTB9'), (u'D3', u'PTA1'), (u'D4', u'PTB23'), (u'D5', u'PTA2'), (u'D6', u'PTC2'), (u'D7', u'PTC3'), (u'D8', u'PTA0'), (u'D9', u'PTC4'), (u'D10', u'PTD0'), (u'D11', u'PTD2'), (u'D12', u'PTD3'), (u'D13', u'PTD1'), (u'D14', u'PTE25'), (u'D15', u'PTE24'), (u'I2C_SCL', u'D15'), (u'I2C_SDA', u'D14'), (u'A0', u'PTB2'), (u'A1', u'PTB3'), (u'A2', u'PTB10'), (u'A3', u'PTB11'), (u'A4', u'PTC10'), (u'A5', u'PTC11'), (u'DAC0_OUT', u'0xFEFE')]))]) is not valid under any of the given schemas
```

autopulated, 2015-09-23:
Ah, the schema says that values can't start with numbers, amongst other things. That's a hangover from original thoughts that things in config would have to be able to be valid symbols, which is no longer the case.

bremoran, 2015-09-23:
Maybe they just need to be valid R-Values?

autopulated, 2015-09-24:
I think allowing anything is reasonable at this point. yotta might just need to do some smarter escaping when defining them in CMake/preprocessor definitions.

commit 786c6508 (2015-09-24): relax the schema on config data to allow any string as a value (fixes #425)
commit f10a1569 (2015-09-24): Merge pull request #427 from ARMmbed/schema-fix

relax the schema on config data to allow any string as a value (fixes #425)
closed 2015-09-24 by autopulated