ARMmbed/yotta #432: if dependency has custom CMakeLists.txt file its test direcrory, then that is alwasy build

jupe, 2015-09-25:
This blocks now cpputest testing in x86 targets, when module need dependencies with custom CMakeLists.txt -file.

```
git clone git@github.com:ARMmbed/mbed-client-cli.git
cd mbed-client-cli/
git checkout -b origin/unittests
yt test  --> fail
mv yotta_modules/mbed-client-trace/test/ yotta_modules/mbed-client-trace/_test
yt test  --> success
````

> while the tests of dependencies aren't normally built, if the dependency has a custom CMakeLists.txt > file in its test directory, then that is always built

autopulated, 2015-09-25:
All the cases of this that I can reproduce are fixed by #428, I'm not sure if that definitely resolves what you report though – so please re-test with master!

autopulated, 2015-09-25:
This should be fixed in yotta 0.8.0, please let me know if you're still experiencing issues!

jupe, 2015-09-25:
now `yt build` build with same scenarous :/

jupe, 2015-09-28:
looks that `yt build mbed-client-cli` -works ! as well as `yt test mbed_client_cli_test`.

autopulated, 2015-09-28:
@jupe Ok cool! Can I mark this as resolved now then?

autopulated, 2015-10-01:
closing this as presumed fixed (I can no longer reproduce it after #428). If this problem recurs please re-open!

closed 2015-10-01 by autopulated

kuggenhoffen, 2015-10-05:
This is still broken. Trying to build ARMmbed/mbed-client-testapp:

git clone ARMmbed/mbed-client-testapp
yt target x86-linux-native
yt build

-> fails on 
Building CXX object ym/mbed-client-trace/test/CMakeFiles/mbed_client_trace_test.dir/Test.cpp.o
and
Building CXX object ym/mbed-client-cli/test/CMakeFiles/mbed_client_cli_test.dir/Test.cpp.o

Both missing "mbed-cpputest/CppUTest/TestHarness.h" which both modules have in their testTargetDependencies

autopulated, 2015-10-05:
The problem that's originally described is fixed. From your emails you have a slightly different problem: please could you describe that in a new issue?

kuggenhoffen, 2015-10-05:
Well, it is still trying to build a dependency's (eg. mbed-client-cli in this case) tests, is this not the same issue?

autopulated, 2015-10-05:
Ok, got it: please see https://github.com/ARMmbed/mbed-client-trace/issues/8 for the solution to that problem. The docs also need updating to explain that.