ARMmbed/yotta #441: Yotta fails to build for x86-osx-native target

yulis, 2015-10-01:
Failed to build https://github.com/ARMmbed/helloyotta for x86-osx-native target.

##### Environment:
- yotta version 0.8.1
- OS X Yosemite version 10.10.5
- XCode version 7.0.1
- Command Line Tools (OS X 10.10) for Xcode version 7.0

```
info: generate for target: x86-osx-native 0.0.7 at /Users/yslob/Projects/my/mbed/helloyotta/yotta_targets/x86-osx-native
-- The C compiler identification is AppleClang 7.0.0.7000072
-- The CXX compiler identification is AppleClang 7.0.0.7000072
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/yslob/Projects/my/mbed/helloyotta/build/x86-osx-native
[4/4] Linking C executable source/helloyotta
FAILED: : && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -O2 -g -DNDEBUG -Wl,-search_paths_first -Wl,-headerpad_max_install_names  source/CMakeFiles/helloyotta.dir/Users/yslob/Projects/my/mbed/helloyotta/source/hello.c.o  -o source/helloyotta  ym/simplelog/source/libsimplelog.a && :
Undefined symbols for architecture x86_64:
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed ```

0xc0170, 2015-10-01:
Hi, thanks for reporting. This issue is related to https://github.com/ARMmbed/helloyotta. Please report it there.

Looking at module.json file, we can see there is target dependency for mbed (there's a module within mbed dependencies where is main symbol defined, for instance).

autopulated, 2015-10-01:
Closing as this is a helloyotta issue – the problem is that helloyotta is now targeting mbed OS by default (where `app_start` is the entry point, instead of `main`). Obviously helloyotta should also work natively, so this is a bug: it should use the correct entry point on different platforms. Please open an issue here: https://github.com/ARMmbed/helloyotta

closed 2015-10-01 by autopulated

jpsecher, 2017-09-22:
For others that are late to the party:

Get helloyotta thus:

    $ git clone git@github.com:0xc0170/helloyotta.git

Add the following to `helloyotta/source/hello.c`:

	int main (int argc, char *argv[]) {
		app_start(argc, argv);
		return 0;
	}