ARMmbed/yotta #442: Could not install public module

toyowata, 2015-10-01:
When I install the ble module (public repo), I have got different results by two PCs (both are Windows 7 and same version of tool installed such as yotta, git etc).

I got module repository error in one PC:

```
E:\tmp>yotta install ble
info: get versions for ble
Traceback (most recent call last):
  File "C:\Python27\Scripts\yotta-script.py", line 9, in <module>
    load_entry_point('yotta==0.8.1', 'console_scripts', 'yotta')()
  File "C:\Python27\lib\site-packages\yotta\main.py", line 239, in main
    status = args.command(args, following_args)
  File "C:\Python27\lib\site-packages\yotta\install.py", line 63, in execCommand
    return installComponent(args)
  File "C:\Python27\lib\site-packages\yotta\install.py", line 194, in installComponent
    working_directory = path
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 352, in satisfyVersion
    return satisfyVersionByInstalling(name, version_required, working_directory, type=type)
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 294, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "C:\Python27\lib\site-packages\yotta\lib\access.py", line 108, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "C:\Python27\lib\site-packages\yotta\lib\registry_access.py", line 461, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "C:\Python27\lib\site-packages\yotta\lib\registry_access.py", line 102, in wrapped
    return fn(*args, **kwargs)
  File "C:\Python27\lib\site-packages\yotta\lib\registry_access.py", line 251, in _listVersions
    'Check that the name is correct, and that it has been published.') % (name, namespace)
yotta.lib.access_common.Unavailable: ble does not exist in the modules registry. Check that the name is correct, and that it has been published.

E:\tmp>
```

But, no error in another PC.

```
D:\tmp>yotta install ble
info: get versions for ble
info: download ble@0.4.8 from https://yotta-private.herokuapp.com
info: get versions for frdm-k64f-gcc
info: download frdm-k64f-gcc@0.1.2 from the public module registry
info: get versions for mbed-gcc
info: download mbed-gcc@0.1.1 from the public module registry
info: get versions for mbed-drivers
info: download mbed-drivers@0.9.0 from the public module registry
info: get versions for mbed-hal
info: download mbed-hal@0.6.4 from the public module registry
info: get versions for cmsis-core
info: download cmsis-core@0.2.7 from the public module registry
info: get versions for ualloc
info: download ualloc@0.2.0 from the public module registry
info: get versions for minar
info: download minar@0.8.1 from the public module registry
info: get versions for core-util
info: download core-util@0.2.1 from the public module registry
info: get versions for compiler-polyfill
info: download compiler-polyfill@1.0.4 from the public module registry
info: get versions for mbed-hal-freescale
info: download mbed-hal-freescale@0.5.2 from the public module registry
info: get versions for mbed-hal-ksdk-mcu
info: download mbed-hal-ksdk-mcu@0.5.7 from the public module registry
info: get versions for uvisor-lib
info: download uvisor-lib@0.7.25 from the public module registry
info: get versions for mbed-hal-k64f
info: download mbed-hal-k64f@0.3.6 from the public module registry
info: get versions for mbed-hal-frdm-k64f
info: download mbed-hal-frdm-k64f@0.4.4 from the public module registry
info: get versions for cmsis-core-freescale
info: download cmsis-core-freescale@0.1.4 from the public module registry
info: get versions for cmsis-core-k64f
info: download cmsis-core-k64f@0.1.5 from the public module registry
info: get versions for dlmalloc
info: download dlmalloc@0.0.6 from the public module registry
info: get versions for minar-platform
info: download minar-platform@0.3.4 from the public module registry
info: get versions for minar-platform-mbed
info: download minar-platform-mbed@0.1.5 from the public module registry

D:\tmp>
```
Not sure why ble module are downloaded from https://yotta-private.herokuapp.com.

Any suggestion?

jbmpark, 2015-10-01:
I've gotten a different error where yotta cannot resolve the module name 'ble'.
Please refer to the below logs and let me get any advice : 

> yotta build
info: get versions for nordic-nrf51822-16k-gcc
info: download nordic-nrf51822-16k-gcc@0.0.9 from the public module registry
info: get versions for ble
error: Failed to satisfy dependencies of E:\work\mbed\ble_nordic\ble-nrf51822:
error: ble does not exist in the modules registry. Check that the name is correct, and that it has been published.
error: ble not available
error: Missing dependencies prevent build. Use `yotta ls` to list them.

autopulated, 2015-10-01:
`ble` isn't the name of a module that currently exists – are there instructions somewhere that guided you to try to use that?

rgrover, 2015-10-01:
the `ble` module might be limited to the private registry.
There's a github/armmbed/ble-examples repo showing how to build ble examples.

closed 2015-10-01 by autopulated

jbmpark, 2015-10-02:
'ble-nrf51822'  has dependency on the 'ble' module.

In the module.json file of 'ble-nrf51822' : 
  "dependencies": {
    "ble": "~0.4.8"
  },

So, I think the 'ble' must be a module, and, without it, 'ble-nrf51822' can't be built.
Is it something I'm missing?

rgrover, 2015-10-02:
yes, the BLE_API respository on github is the 'ble' module.