ARMmbed/yotta #451: add yotta validate command

jrobeson, 2015-10-05:
yotta validate would make sure of things like:
  * license is specified correctly
  * spit out json formatting errors
  * make sure the module name is specified correctly
  * other stuff I'm sure people will chime in on

Composer has a similar command.

I thought about this after seeing this:

```bash
$ yotta config
warning: test has invalid module.json:
warning:   name value u'Test' does not match u'^[a-z]+[a-z0-9-]*$'
```

autopulated, 2015-10-05:
Your module/target description is validated by yotta whenever its read (by practically every subcommand) – so a simple `yotta ls`, `yotta config` like you discovered) will do this.

I'm guessing that you ended up with an invalid module name after running `yotta init`? (In which case it sounds like `yotta init` needs to be careful checking its input.)

(Note also that in yotta 0.8 onwards, the name "test" should be explicitly checked for and warned about, because it can cause all sorts of problems with confusion with the "test" directory ;)

jrobeson, 2015-10-05:
I actually renamed it to  Test because the initial name was dumber.

ilg-ul, 2015-12-02:
It might also spot deprecated syntax, for example uses of `licenses` instead of `license`.