ARMmbed/yotta #453: yt link - traceback if there's no module linked with that name

0xc0170, 2015-10-05:
Yotta v0.8.0, OS: windows

```
λ yt link mbed-hal-st                                                                                                                      
warning: C:\xxx\mbed-hal-st -> C:\Python27\Lib\yotta_modules\mbed-hal-st -> BROKEN 
Traceback (most recent call last):                                                                                                         
  File "C:\Python27\Scripts\yt-script.py", line 9, in <module>                                                                             
    load_entry_point('yotta==0.8.0', 'console_scripts', 'yt')()                                                                            
  File "C:\Python27\lib\site-packages\yotta-0.8.0-py2.7.egg\yotta\main.py", line 239, in main                                              
    status = args.command(args, following_args)                                                                                            
  File "C:\Python27\lib\site-packages\yotta-0.8.0-py2.7.egg\yotta\link.py", line 54, in execCommand                                        
    fsutils.symlink(src, dst)                                                                                                              
  File "C:\Python27\lib\site-packages\yotta-0.8.0-py2.7.egg\yotta\lib\fsutils.py", line 94, in symlink                                     
    _symlink(source, link_name)                                                                                                            
  File "C:\Python27\lib\site-packages\yotta-0.8.0-py2.7.egg\yotta\lib\fsutils_win.py", line 47, in _symlink                                
    junction.create(source, link_name)                                                                                                     
  File "C:\Python27\lib\site-packages\ntfsutils-0.1.3-py2.7.egg\ntfsutils\junction.py", line 116, in create                                
    raise Exception("%s is not a directory" % source)                                                                                      
Exception: C:\Python27\Lib\yotta_modules\mbed-hal-st is not a directory                                                                    
```

Can we pinpoint there's no linked module with that name, instead of a traceback?

bogdanm, 2015-10-05:
There's a more fundamental problem here. On Linux, `yt link` will give you the `BROKEN` message, but the link is still created, so if you later do `yt link` in `mbed-hal-st`, the "broken" link that you created earlier will work. This doesn't happen on Windows: after you get the `BROKEN` message, the link is not created. Not sure if the Linux behaviour can be reproduced in Windows.

autopulated, 2015-10-05:
+1 thanks for reporting!

autopulated, 2015-10-05:
@bogdanm I don't think it's possible with directory junctions :(

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-891

PR #636 (2015-12-18): Don't crash if creating a link fails
 * produce a useful error in the case that the second half failed because the
   first half hasn't been completed (this happens on windows)
 * Otherwise reduce the exception encountered to an error message.

Fixes #453
closed 2015-12-18 by autopulated