ARMmbed/yotta #465: Module - yt version command changes line endings

0xc0170, 2015-10-06:
OS: windows yotta version: 0.8.0

Most/all of module.json file have unix line endings, every time I do yt version, it changes the file to system's default line endings (I assume this happens as I am getting on windows machine windows line endings :tired_face:) which is annoying. Is there a way to configure this?

autopulated, 2015-10-07:
There's no way to configure this: whenever yotta writes a json file it'll use system default line endings (that seems like the right behaviour to me?).

You should probably have git set up to clone things with native line endings, and do the conversion to unix line endings internally (the `git config core.autocrlf` setting).

autopulated, 2015-10-07:
(leaving this open as a bug: yotta-should-preserve-existing-line-endings-when-modifying-json-files)

0xc0170, 2015-10-07:
:+1: with the above

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-890