ARMmbed/yotta #467: yotta access to git hosted yotta target

markus-becker-tridonic-com, 2015-10-06:
I am trying to yotta install in a yotta module, which depends on a yotta target which is not yet upstreamed, but in my own repository.

This is .yotta.json:
```
{
  "build": {
    "target": "st-nucleo-f091rc-gcc,https://github.com/markus-becker-tridonic-com/st-nucleo-f091rc-gcc.git"
  }
}
```

When then doing
```
$ yotta install
info: get versions for st-nucleo-f091rc-gcc
info: failed to fetch Github object, re-trying with authentication...

You need to log in with Github to do this.
Press enter to continue.
Your browser will open to complete login.
waiting for response...
If you are unable to use a browser on this machine, please copy and paste this URL into a browser:
http://yottabuild.org/?provider=github#...
```
and opening the website and doing the usual OAuth stuff, I end up with the following message:
```
Sorry, something went wrong.
Must share at least one verified email to merge accounts
```

markus-becker-tridonic-com, 2015-10-06:
Interestingly, this .yotta.json works:
```
{
  "build": {
    "target": "st-nucleo-f091rc-gcc,markus-becker-tridonic-com/st-nucleo-f091rc-gcc"
  }
}
```

closed 2015-10-07 by markus-becker-tridonic-com

autopulated, 2015-10-07:
Currently user/repo is the canonical way of specifying something in github – URLs to github pages should probably be automatically transformed to the right form though.