ARMmbed/yotta #468: Proxy environment variables are ignored for GitHub access method

markus-becker-tridonic-com, 2015-10-07:
We're trying to use yotta on Windows behind a corporate proxy.

We are setting the respective environment variables:
```
set http_proxy=http://8.8.8.8
set https_proxy=http://8.8.8.8
set HTTP_PROXY=http://8.8.8.8
set HTTPS_PROXY=http://8.8.8.8
```

yotta is picking this up for access to the yotta registry, but not for access to github. As far as I understand, for the registry the Python requests module is used, which handles this just fine. For access to github, however, the PyGitHub module is used, which only deals with the proxy since https://github.com/PyGithub/PyGithub/commit/7c79c1b80b8f426806440902753547de335ed71f
which does not seem to be included in a release, yet.

I was able to patch it into the released version of PyGitHub, so that it works:
```
    def __createConnection(self):
        kwds = {}
        if not atLeastPython3:  # pragma no branch (Branch useful only with Python 3)
            kwds["strict"] = True  # Useless in Python3, would generate a deprecation warning
        if atLeastPython26:  # pragma no branch (Branch useful only with Python 2.5)
            kwds["timeout"] = self.__timeout  # Did not exist before Python2.6
        #return self.__connectionClass(self.__hostname, self.__port, **kwds)

        ##
        ## Connect through a proxy server with authentication, if http_proxy
        ## set.
        ## http_proxy: http://user:password@proxy_host:proxy_port
        ##
        proxy_uri = os.getenv('http_proxy') or os.getenv('HTTP_PROXY')
        if proxy_uri is not None:
            url = urlparse.urlparse(proxy_uri)
            conn = self.__connectionClass(url.hostname, url.port, **kwds)
            headers = {}
            if url.username and url.password:
                auth = '%s:%s' % (url.username, url.password)
                headers['Proxy-Authorization'] = 'Basic ' + base64.b64encode(auth)
            conn.set_tunnel(self.__hostname, self.__port, headers)
        else:
            conn = self.__connectionClass(self.__hostname, self.__port, **kwds)
        return conn
```

The other quick fix for me was to patch sourceparse.py of yotta from
```
    elif parsed.netloc.endswith('github.com'):
        # any URL onto github should be fetched over the github API, even if it
        # would parse as a valid git URL
        return VersionSource('github', parsed.path, parsed.fragment)
```
to
```
    elif parsed.netloc.endswith('github.com'):
        # any URL onto github should be fetched over the github API, even if it
        # would parse as a valid git URL
        return VersionSource('git', without_fragment, parsed.fragment)
```

autopulated, 2015-10-07:
OK thanks for reporting – upgrading to PyGithub 2 is the best solution for this.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-889

mlnx, 2016-01-13:
@markus-becker-tridonic-com 
Have you upgraded to PyGithub 2? If so, is the problem solved?

autopulated, 2016-01-13:
To be clear, yotta needs to update to PyGithub2, it isn't something a user can do. PyGithub v2 is still in alpha development, and isn't suitable for yotta to use just yet (http://pygithub.github.io/PyGithub/v2/index.html)

bogdanm, 2016-01-13:
Alternatively, try using socksify (http://linux.die.net/man/1/socksify) or a similar program with yotta, it might fix your issue.

autopulated, 2016-01-14:
:+1: would be interested to know if socksify works!

markus-becker-tridonic-com, 2016-01-25:
PyGitHub 1.26.0 fixed this: https://github.com/PyGithub/PyGithub#version-1260-november-5th-2015

closed 2016-01-25 by markus-becker-tridonic-com

autopulated, 2016-01-25:
Hadn't noticed that – thanks for the update!