ARMmbed/yotta #473: Provide a framework for analytics data to be captured and sent to the registry

hugovincent, 2015-10-08:
There are a number of types of analytics that would be useful to collect. Proposal:
* collect and upload anonymous analytics by default, with simple opt-out.
   * host OS, python version etc
   * toolchain (armcc, gcc, etc) and debugger choices
   * modules being built and the target, but only include targets/modules that are published to the registry and/or not marked as private.

For discussion:
* optionally upload errors/crashlogs?
* code size. See also @jupe's idea in #407.

autopulated, 2015-10-08:
crash logs would be very useful. Do you know of any good python frameworks for doing that?

autopulated, 2015-10-15:
Hello @mindw @PrzemekWirkus @0xc0170 @bogdanm @sg- @stephenpaulger @thegecko @stevenewey @danros – tagging you because I think you might have come across python client-side analytics and/or crash reporting frameworks. Are there any you can recommend?

stevenewey, 2015-10-15:
getsentry.com is great for logging and alerting on exceptions. I've only used it in server side projects. I've got a trial account up and running for Workshop at the moment that I can add you to if you want to try it out.