ARMmbed/yotta #487: Markup on Pypi site not properly formatted

cmonr, 2015-10-13:
Whether intended or not, the markup does not resolve properly when viewed on the Pypi site: https://pypi.python.org/pypi/yotta/0.8.3

posborne, 2015-10-14:
This should be fixable using something like setuptools-markdown (https://github.com/msabramo/setuptools-markdown).  Unfortunately, there is an issue I am working with upstream to fix in pypandoc that prevents this from being a suitable solution at the moment (require pandoc install on all clients, not just maintainer machine).

Relevant Tickets:
* https://github.com/msabramo/setuptools-markdown/issues/6
* https://github.com/bebraw/pypandoc/issues/70

If we wanted to do something else in the short term, you can use this code as a reference.  Not as nice as setuptools-markdown but works: https://github.com/digidotcom/python-suitcase/blob/master/setup.py#L10

autopulated, 2015-10-14:
TBH It might be best to simply maintain separate docs for the PyPI page and github readme. They can be tailored specifically to the audience then too (i.e. the PyPI one can say things like report bugs on github...)

posborne, 2015-10-14:
That seems fair.  I was able to push a patch through and there is a new version of pypandoc released now that resolves the issues I mentioned previously, so use conversion of the readme may be a possibility as well now.   That being said, tailoring could make sense.

autopulated, 2015-10-14:
see #488 ;)

closed 2015-10-14 by autopulated