ARMmbed/yotta #496: Ability to publish, search, etc. to private repository

beriberikix, 2015-10-14:
It would be great to use all the features built into the exec to use a private repo. Even without open-sourcing (#452,) it would be nice to use the existing commands to push, get build versions, etc. I imagine a lot of that is just documenting a repo structure `yt` expects and exposing a .config to point the exec to the private repo.

autopulated, 2015-10-15:
Thanks for the feedback about this – it's something that we might well enable in the future (in a similar way that npm has previously provided a private registry service, and now provides access-controlled modules within their main registry).

We don't have an immediate plans or a timescale for this yet though.

For now it might be useful to use git/hg/Github references in your dependencies to pull things from github. You can still use version specifications with these URLs, for example:

```JSON
  "dependencies":{
      "some-private-module": "ARMmbed/some-private-module#~1.2.3"
  }
```

Will depend on tagged versions matching ~1.2.3. (Be sure to do `git push --tags` after using `yotta version` so that your tags are pushed to the git remote.

There's a little more info here: http://yottadocs.mbed.com/tutorial/privaterepos.html

beriberikix, 2015-10-15:
Thanks for the reply, would be a great addition. There are many other package managers that both let you specify custom repositories (not just packages per project) and even host the whole thing. Examples:

* [CocoaPods (iOS)](https://guides.cocoapods.org/making/private-cocoapods.html)
* [Bower (js)](https://github.com/bower/registry)
* [Gradle (Android)]( https://docs.gradle.org/current/userguide/artifact_dependencies_tutorial.html#N10621)
* [PyPi (Python)](https://packaging.python.org/en/latest/self_hosted_repository/)
* ...probably half-dozen more

I have a fair bit of experience using package managers :smile:

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSFW-888

Overdrivr, 2016-02-22:
I second Gradle, that is more than just a package manager, but an immensely powerful automation tool with first-class support for build systems. It is not tied at all to Android. You can build pretty much any language.

Did you guys at ARM yotta evaluated this tool rather than going for a CMAKE based build ? Because I truly believe it would make your life a lot more easier.

Just by declaring a couple of tasks you can configure a project build that directly fetches dependencies by cloning other git repositories, checkout the latest tags, configure the build, build, run tests, move files accross folder hierarchy, package and distribute is just amazing. And just that from a single command in the cli and 50 lines of code in a build.gradle file.
It also makes the build highly reproducible, and, extra bonus, this is 100% compatible with Continouus Integration.

PS: I don't work for Gradle.