ARMmbed/yotta #497: Cannot start debugging in Windows using valinor 0.0.10

toyowata, 2015-10-15:
After build example-mbedos-blnky project with frdm-k64f-gcc target, I launched the uVIsion debugger using following command. It works fine.

```
E:\tmp\mbed_OS\example-mbedos-blinky>valinor --target K64F -t uvision -d build\frdm-k64f-gcc\source build\frdm-k64f-gcc\
source\example-mbedos-blinky
```

However, 'yt debug xxx' command doesn't work.

```
E:\tmp\mbed_OS\example-mbedos-blinky>yt debug example-mbedos-blinky
info: found example-mbedos-blinky at source\example-mbedos-blinky
Traceback (most recent call last):
  File "C:\Python27\Scripts\valinor-script.py", line 9, in <module>
    load_entry_point('valinor==0.0.10', 'console_scripts', 'valinor')()
  File "C:\Python27\lib\site-packages\valinor\main.py", line 83, in main
    ide_tool = ide_detection.select(available_ides, args.target, project_settings)
  File "C:\Python27\lib\site-packages\valinor\ide_detection.py", line 126, in select
    if targets.Targets(project_settings.get_env_settings('definitions')).is_supported(target, ide):
  File "C:\Python27\lib\site-packages\project_generator\targets.py", line 41, in __init__
    self.targets = [ splitext(f)[0] for f in listdir(target_dir) if isfile(join(target_dir,f)) ]
WindowsError: [Error 3] 指定されたパスが見つかりません。: 'C:\\Users\\toywat01\\.pg\\definitions\\target/*.*'
error: debug process exited with status 1
```

It says ```The system cannot find the path specified: 'C:\\Users\\toywat01\\.pg\\definitions\\target/*.*```.
I think the .pg directory is created by pgen (progject_generator), but there is no ```C:\\Users\\toywat01\\.pg``` direcotry in my host PC.

The #389 looks same issue, but I still have a problem with latest revision (valinor v0.0.10).
Any suggestion?

toyowata, 2015-10-15:
```
E:\tmp\mbed_OS\example-mbedos-blinky>yt --version
0.8.5

E:\tmp\mbed_OS\example-mbedos-blinky>pgen --version
0.7.4

E:\tmp\mbed_OS\example-mbedos-blinky>valinor --version
0.0.10
```

0xc0170, 2015-10-15:
Thanks for reporting. I'll send a patch to progen within minutes.

0xc0170, 2015-10-15:
Published a fix to progen, which should fix this issue.

autopulated, 2015-10-15:
OK awesome, thanks!

closed 2015-10-15 by autopulated