ARMmbed/yotta #503: `npm shinkwrap` equivalent

autopulated, 2015-10-19:
To release a stable version of an application it's simple to commit the `yotta_modules` folder. If you want to release a stable version of a module, this approach will only work as long as other modules being included in the system don't first pull in other versions of the dependencies required.

While that's unavoidable with the non-iterative dependency solver that yotta currently uses, it's desirable that this case produces an explicit warning or error that the dependencies of the stable module have been overridden.

Respecting `yotta-shrinkwrap.json` files in dependencies would provide a means to do this. It would also be compatible with any future dependency solver that tries to iteratively optimise dependency versions.

(in npm, the `npm-shrinkwrap.json` format is actually the same as `npm ls --json` output)

bogdanm, 2015-10-19:
(for reference: https://docs.npmjs.com/cli/shrinkwrap)

So, if I got this right, this would be the workflow:

1. create a `release-xxx` branch on each release repository. That branch will have the `npm-shrinkwrap.json` file in it.
2. if you want to use a module from that release later, you'll depend on (say) `ARMmbed/mbed-drivers#release-xxx`, which in turn will bring in all the versions at the moment of the release.

That sounds quite reasonable to me.

bremoran, 2015-10-19:
Agreed.  This seems reasonable to me.  I just want to make sure we get the ```ls --json``` format right so that it caters to many possible use-cases.

autopulated, 2015-10-19:
@bogdanm, yup, you could also publish the shrinkwrap file if we wanted specific published versions with pegged dependencies.

bogdanm, 2015-10-19:
@autopulated, I thought about that too, but I couldn't make sense of versioning. So let's say that I have my module at version `x.y.z`, I add my shrinkwrap file to it, commit, then I want to publish this new version of my module . What kind of version should I publish (major/minor/patch), keeping in mind that I don't want to break `^` and `~' dependency specs on my module? The only reasonable thing I could come up with was `x.y.z-release-xxx`, but that doesn't look compatible with the way we do things.

bremoran, 2015-10-19:
I didn't think we ever updated releases...  Isn't that the whole point of this shrinkwrap process?

bremoran, 2015-10-19:
I don't think we should publish modules with shrinkwrap files in them, because they defeat yotta's normal use of versioning.

I also expect shrinkwrap to be used on top-level modules, such as examples.  It doesn't make sense to me to use it for anything else.  Executables shouldn't be in the registry, so they wouldn't be published anyway.

bogdanm, 2015-10-19:
@bremoran, what if I need to depend on the `sockets` module from a supported release of mbed OS? That's a legitimate use case and a good reasons to have a module shrinkwrapped.

bremoran, 2015-10-19:
@bogdanm: we need to work out what the semantics for doing that are.  There's an argument for publishing `x.y.z-release-xxx` modules, which have pegged dependencies.  The problem with shrinkwrap files is that the expect a particular target.  `sockets` (indirectly) depends on different things, depending on your yotta-config.

bogdanm, 2015-10-19:
@bremoran, yes, we definitely need to figure out how to properly include targets in the equation, The shrinkwrap file should have the dependencies for all acceptable targets at the moment of the release, plus (probably) the actual target versions used at the moment of the release.

bremoran, 2015-10-19:
Here's an option: the mbed-os module could have a shrinkwrap file that contains every module in the release, but yotta would need to ignore modules in the shrinkwrap file that aren't in your dependency list.

That should deal with the problem pretty effectively.

autopulated, 2015-10-19:
@bogdanm you could do a patch. Then do another patch for the non-frozen release.

@bremoran yotta would have to ignore modules in the shrink-wrap that aren't in dependencies, because the shrink-wrap will be the same for all targets.

bremoran, 2015-10-19:
@autopulated: Yes, that's what I meant.  If we could make depending on a module that contains a shrinkwrap file implicitly apply the shrinkwrap to the whole tree, that would make it possible to depend on a release just by depending on a particular version of the release repository.  I think that makes a pretty clean user experience.

autopulated, 2015-10-19:
@bremoran it would, although see my caveat in the original issue: in some cases it will (in simplest implementation) fail with an error if a different version already exists.

hugovincent, 2015-10-19:
Out of interest:
>  It would also be compatible with any future dependency solver that tries to iteratively optimise dependency versions.

See this http://rox.sourceforge.net/desktop/node/954.html on "Adapting the OPIUM solver" ... might be interesting. Not necessarily surprising, but it turns out this future dependency solver might be very simple to express as a SAT problem, and then should be fast (even as pure python). This post explains an approach: http://thread.gmane.org/gmane.comp.file-systems.zero-install.devel/3082

autopulated, 2015-10-19:
@hugovincent yup that looks really helpful, thanks!

jrobeson, 2015-10-20:
oops. I asked the question on #508 when I should have asked it here. Is there any benefit of the npm shrinkwrap process over composer or bundler's lock file concept? and always locking versions for applications by default.  Thus yotta update regenerates yotta.lock with the versions of the application with no extra command to remember to run.

autopulated, 2015-10-20:
@jrobeson the benefit of this is that it can be potentially used with modules: we're experimenting with it as a way of preserving official tested combinations of the mbed OS modules for releases.

If you're developing an *application* (as opposed to a re-usable module/set of modules), it's perfectly reasonable to commit your yotta_modules (locking your dependencies unless you specifically go and update, and giving you 100% visibility of what changes). You could also use shrink-wrap for this, but I don't think it should be the default, as we want to encourage people to live on the latest versions of things (especially at this very early stage).

jrobeson, 2015-10-20:
both npm and bower still cause problems for me to this very day due to the lack of a lockfile, thus my concern. I hope a lock file is looked into at some point though.

autopulated, 2015-10-20:
Ok, thanks for the feedback! – could you explain more about what a lockfile does for you that a shrink-wrap wouldn't (or committing your dependencies), is it just the fact that it wouldn't be a default workflow?

jrobeson, 2015-10-20:
default is the biggest problem yes and I don't like committing my dependencies either. 

When I use composer (I'm mostly a web app dev) I end up with with all my dependencies pulled directly into `vendor/package-namespace/package-name` (where package-namespace is often a company) locked to the specific version that is set in the lockfile. That way I never have to worry if I'll ever get the same version of the code again, whether it's library versioned by semver or even linked to a specific vcs commit.  I choose when to update my libraries by running composer update (or composer update package-name)

I consider the lack of a lockfile to be npm/bower biggest missing feature, especially since that ecosystem tends to move incredibly fast.

autopulated, 2016-01-19:
Fixed by #508

closed 2016-01-19 by autopulated