ARMmbed/yotta #506: yotta installation fails on mac os x el capitan due to rootless mode

LiyouZhou, 2015-10-19:
User does not have right to folder /usr/local causing installation to fail. Unless disable rootless mode, the permission to system folders cannot be changed.

autopulated, 2015-10-19:
Thanks for reporting this. To clarify, is this during `pip install`, or when you try to run yotta?

If you could include the full backtrace/error message that'd be really helpful.

LiyouZhou, 2015-10-19:
Unfortunately terminal printout lost. If I can find a fresh machine, I will try to reproduce. It was during the installation. Also I had to install cython in order to successfully install yotta.

0xc0170, 2015-10-20:
@LiyouZhou I have just published pyOCD 0.6.0 which removes the requirement on cython.

jupe, 2016-03-23:
I think I faced this same issue, cannot install yotta to my MAC.

autopulated, 2016-03-24:
@jupe OK thanks for reporting – what error message do you see?