ARMmbed/yotta #511: Retry failed extractions of tar files

andresag01, 2015-10-20:
After modifying the `module.json` to update the version for the BLE dependency, I run `yotta update` and got the following error message. Seems like there was a problem trying to extract the tar file downloaded.

```
$ yotta update
info: get versions for nrf51dk-gcc
info: get versions for nordic-nrf51822-32k-gcc
info: get versions for mbed-gcc
info: get versions for ble
info: download ble@1.0.0 from the public module registry
error: failed to extract tarfile CRC check failed 0x57ad2cd8 != 0x70b59344L
Traceback (most recent call last):
  File "/usr/local/bin/yotta", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/site-packages/yotta/main.py", line 239, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta/update.py", line 28, in execCommand
    updateDeps(args)
  File "/usr/local/lib/python2.7/site-packages/yotta/update.py", line 58, in updateDeps
    test = update_test_deps
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 554, in satisfyDependenciesRecursive
    test = test
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 349, in __getDependenciesRecursiveWithProvider
    test = test
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 249, in __getDependenciesWithProvider
    satisfyDep, specs
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 231, in satisfyDep
    self.getName()
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 540, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 297, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 312, in _satisfyVersionByInstallingVersion
    'Dependency "%s":"%s" is not a valid %s.' % (name, version_required, type)
Exception: Dependency "ble":"^1.0.0" is not a valid module.
```

autopulated, 2015-10-20:
That's an interesting failure – if you remove `yotta_modules/ble` and try again does it work?

andresag01, 2015-10-23:
To solve the problem we removed the entire `yotta_modules` directory. However, it will probably work if you just remove the `yotta_modules/ble` and any temporary directories in `yotta_modules`.