ARMmbed/yotta #513: Display current version of yotta and suggest upgrade

mlnx, 2015-10-21:
We should be listing the version of yotta every time we invoke the yotta command (this helps at providing support when there is an issue).

As a bonus, we could display that a new version of yotta is available (maybe after a long build and if something goes wrong?)

autopulated, 2015-10-21:
No we shouldn't, this would get really irritating.

We should display notifications when a new version of yotta is available, there's already an issue for that. Please ask people to report `yotta --version` when reporting an issue, this is a fairly natural thing to do.

(closing this as the valid suggestion is a duplicate)

closed 2015-10-21 by autopulated

mlnx, 2015-10-21:
For the first request "display yotta version every time the yotta command is invoked", this will *definitely* help everybody as will *reduce* the iteration in the communication (not having to come back and ask for this information and wait for the reply). From my experience at providing support, this will be an important improvement.

I'm referring to having a *single* and *short* line straight after yotta is invoked, for example:
``` Yotta version: Y.X.Z```

Please re-consider this feature.