ARMmbed/yotta #514: Weird characters when using yotta under Linux

mlnx, 2015-10-21:
Ubuntu 14.04.2 LTS
Yotta version: 0.8.5

Example
```
marsal01@xxx:~/mbed-client-linux-example$ yt ls
mbed-client-linux-example 1.0.0
ââ mbed-client 1.1.8
  â£â mbed-client-c 1.0.6 yotta_modules/mbed-client-c
  â ââ nanostack-libservice 3.0.6 yotta_modules/nanostack-libservice
  â£â mbed-client-linux 1.0.2 yotta_modules/mbed-client-linux
  ââ mbed-client-mbedtls 1.0.6 yotta_modules/mbed-client-mbedtls
    ââ mbedtls 2.1.4 yotta_modules/mbedtls
```

autopulated, 2015-10-21:
Thanks for reporting this, these are UTF-8 unicode characters, which your terminal program isn't displaying correctly.

Please could you share:
 * python version
 * output of `locale` command
 * which terminal program you're using

It might be that yotta needs to check for UTF-8 support in linux terminals, I had assumed UTF-8 would be common enough to just work.

mlnx, 2015-10-21:
Thanks for the quick response. This is the info you requested:

- Python 2.7.6
- locale
```
LANG=en_GB.UTF-8
LANGUAGE=en_GB:en
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
```
- LANG
```en_GB.UTF-8```
- Putty terminal under Windows 7

autopulated, 2015-10-21:
Ah, you're ssh-ing into the linux box?

You need to configure PuTTY to use UTF-8 to match your linux box, as described here: http://thegreyblog.blogspot.co.uk/2009/08/configuring-putty-to-use-utf-8.html

mlnx, 2015-10-21:
Yeap! that solves the issue. Thanks!
Closing the ticket.

closed 2015-10-21 by mlnx

autopulated, 2015-10-21:
awesome, no problem. This should be a good reference for anyone else with the same issue!