ARMmbed/yotta #524: ninja executable is called ninja_build on Fedora Linux v22

meriac, 2015-10-24:
Hi James,

it would be nice to check for **ninja_build** in case **ninja** can't be found. My workaround so far:
```bash
sudo ln -s ninja-build /usr/bin/ninja
```
@autopulated

autopulated, 2015-10-24:
is ninja_build / ninja-build a name that some linux distros use?

autopulated, 2015-10-24:
Ok should have read the title (=yes, fedora v22). Thanks for reporting this.

meriac, 2015-10-26:
@autopulated In case you want to test your changes, you can find a VMware image of a yotta installation on Fedora 22 at http://get.meriac.com/ARMmbed.zip

autopulated, 2015-10-30:
@meriac which version of python ships with fedora v22? (There's a helpful `shutil.which` function in python >=3.3, would be nice to use that)

closed 2015-10-30 by autopulated

meriac, 2015-10-30:
Thanks a lot for the patch!
@autopulated FC22 uses python 2.7

autopulated, 2015-10-30:
no problem, figured it wasn't too hard to detect things even without shutil.which

PR #547 (2015-10-30): add fsutils.which for finding executables; sometimes ninja is called …
…ninja-build, support that; fixes #524