ARMmbed/yotta #527: error: You do not have 'intelhex' installed

jna29, 2015-10-24:
On OS X 10.8.5 (also on 10.9.5), yotta cannot find the intelhex package, even though it is installed.

```
eduroam-244-197:BLE_Beacon jaduol$ yotta build
info: generate for target: blenano-gcc 0.0.2 at /Users/jaduol/ble-examples/BLE_Beacon/yotta_targets/blenano-gcc
GCC version is: 4.9.3
suppressing warnings from ble-nrf51822
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jaduol/ble-examples/BLE_Beacon/build/blenano-gcc
[1/1] Linking CXX executable source/ble-beacon
FAILED: : && /usr/local/bin/arm-none-eabi-g++   -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -mcpu=cortex-m0 -mthumb -T/Users/jaduol/ble-examples/BLE_Beacon/yotta_targets/nordic-nrf51822-16k-gcc/CMake/../ld/NRF51822.ld -Wl,-Map,source/ble-beacon.map -Wl,--start-group source/CMakeFiles/ble-beacon.dir/Users/jaduol/ble-examples/BLE_Beacon/source/main.cpp.o  ym/mbed-hal-nrf51822-mcu/existing/mbed-hal-nrf51822-mcu.a ym/ble/source/ble.a ym/ble-nrf51822/source/ble-nrf51822.a ym/ble/source/ble.a ym/ble-nrf51822/source/ble-nrf51822.a ym/mbed-hal-nrf51822-mcu/existing/mbed-hal-nrf51822-mcu.a ym/mbed-hal/source/mbed-hal.a ym/mbed-hal-nordic/yotta_dummy_lib_mbed_hal_nordic/mbed-hal-nordic.a ym/mbed-hal-blenano/source/mbed-hal-blenano.a ym/mbed-drivers/source/mbed-drivers.a ym/minar/source/minar.a ym/core-util/source/core-util.a ym/minar-platform/yotta_dummy_lib_minar_platform/minar-platform.a ym/minar-platform-mbed/source/minar-platform-mbed.a ym/mbed-hal-nrf51822-mcu/existing/mbed-hal-nrf51822-mcu.a ym/mbed-hal/source/mbed-hal.a ym/mbed-hal-nordic/yotta_dummy_lib_mbed_hal_nordic/mbed-hal-nordic.a ym/mbed-hal-blenano/source/mbed-hal-blenano.a ym/mbed-drivers/source/mbed-drivers.a ym/minar/source/minar.a ym/core-util/source/core-util.a ym/minar-platform/yotta_dummy_lib_minar_platform/minar-platform.a ym/minar-platform-mbed/source/minar-platform-mbed.a ym/compiler-polyfill/yotta_dummy_lib_compiler_polyfill/compiler-polyfill.a ym/ualloc/source/ualloc.a ym/dlmalloc/source/dlmalloc.a -Wl,-wrap,_malloc_r -Wl,-wrap,_calloc_r -Wl,-wrap,_realloc_r -Wl,-wrap,_free_r -Wl,-wrap,_sbrk_r ym/cmsis-core/source/cmsis-core.a ym/cmsis-core-nordic/yotta_dummy_lib_cmsis_core_nordic/cmsis-core-nordic.a ym/cmsis-core-nrf51822/source/cmsis-core-nrf51822.a ym/cmsis-core/source/cmsis-core.a ym/cmsis-core-nordic/yotta_dummy_lib_cmsis_core_nordic/cmsis-core-nordic.a ym/cmsis-core-nrf51822/source/cmsis-core-nrf51822.a  -lstdc++ -lsupc++ -lm -lc -lgcc -lstdc++ -lsupc++ -lm -lc -lgcc -Wl,--end-group  --specs=nano.specs -o source/ble-beacon && cd /Users/jaduol/ble-examples/BLE_Beacon/build/blenano-gcc/source && arm-none-eabi-objcopy -O ihex ble-beacon ble-beacon.hex && python /Users/jaduol/ble-examples/BLE_Beacon/yotta_targets/nordic-nrf51822-16k-gcc/CMake/../scripts/merge_hex.py /Users/jaduol/ble-examples/BLE_Beacon/yotta_targets/nordic-nrf51822-16k-gcc/CMake/../softdevice/s130_nrf51_1.0.0_softdevice.hex ble-beacon.hex ble-beacon-combined.hex && cd /Users/jaduol/ble-examples/BLE_Beacon/build/blenano-gcc/source && /usr/local/bin/arm-none-eabi-objcopy -O binary ble-beacon ble-beacon.bin
error: You do not have 'intelhex' installed. Please run 'pip install intelhex' then retry.
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
eduroam-244-197:BLE_Beacon jaduol$ git branch
* hackathon
eduroam-244-197:BLE_Beacon jaduol$ pip install intelhex
Requirement already satisfied (use --upgrade to upgrade): intelhex in /usr/local/lib/python2.7/site-packages
eduroam-244-197:BLE_Beacon jaduol$ pip list
argcomplete (0.9.0)
cffi (1.2.1)
colorama (0.3.3)
cryptography (1.0.2)
enum34 (1.0.4)
functools32 (3.2.3.post2)
hgapi (1.7.2)
hidapi (0.7.99.post8)
idna (2.0)
intelhex (2.0)
ipaddress (1.0.14)
Jinja2 (2.8)
jsonpointer (1.9)
jsonschema (2.5.1)
MarkupSafe (0.23)
mbed-test-wrapper (0.0.2)
mercurial (3.2.1-20141114)
ndg-httpsclient (0.4.0)
pathlib (1.0.1)
pip (7.1.2)
project-generator (0.7.6)
project-generator-definitions (0.1.3)
pyasn1 (0.1.9)
pycparser (2.14)
PyGithub (1.25.2)
PyJWT (1.4.0)
pyOCD (0.5.1)
pyOpenSSL (0.15.1)
PyYAML (3.11)
requests (2.8.1)
semantic-version (2.4.2)
setuptools (18.3.1)
six (1.10.0)
valinor (0.0.11)
vboxapi (1.0)
wheel (0.26.0)
xmltodict (0.9.2)
yotta (0.8.5)```

jrobeson, 2015-10-24:
can you import intelhex in the python repl ?

autopulated, 2015-10-29:
I'm closing this here as it is not an issue in yotta itself, from your error message it seems `intelhex` is used in a script defined by your target (or one of the targets that it inherits from), and for some reason that import is failing.

Please could you report this as an issue to the author of that target description!

closed 2015-10-29 by autopulated