ARMmbed/yotta #528: Yotta Build command fails in Windows if the directory path contains a space

Hydrogencyanide1, 2015-10-24:
This is the same issue as #133.
```
(workspace) C:\Users\Nick Guin\ble-examples\BLE_Beacon>yt build
info: generate for target: blenano-gcc 0.0.2 at C:\Users\Nick Guin\ble-examples\
BLE_Beacon\yotta_targets\blenano-gcc
GCC version is: 4.9.3
suppressing warnings from ble-nrf51822
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Nick Guin/ble-examples/BLE_Beacon/
build/blenano-gcc
[1/1] Linking CXX executable source/ble-beacon
FAILED: cmd.exe /C "cd . && C:\yotta\gcc\bin\arm-none-eabi-g++.exe   -fno-except
ions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=a
lignment -Wl,-wrap,main -mcpu=cortex-m0 -mthumb -TC:/Users/Nick Guin/ble-example
s/BLE_Beacon/yotta_targets/nordic-nrf51822-16k-gcc/CMake/../ld/NRF51822.ld -Wl,-
Map,source/ble-beacon.map -Wl,--start-group source/CMakeFiles/ble-beacon.dir/C_/
Users/Nick_Guin/ble-examples/BLE_Beacon/source/main.cpp.o  ym/mbed-hal-nrf51822-
mcu/existing/mbed-hal-nrf51822-mcu.a ym/ble/source/ble.a ym/ble-nrf51822/source/
ble-nrf51822.a ym/ble/source/ble.a ym/ble-nrf51822/source/ble-nrf51822.a ym/mbed
-hal-nrf51822-mcu/existing/mbed-hal-nrf51822-mcu.a ym/mbed-hal/source/mbed-hal.a
 ym/mbed-hal-nordic/yotta_dummy_lib_mbed_hal_nordic/mbed-hal-nordic.a ym/mbed-ha
l-blenano/source/mbed-hal-blenano.a ym/mbed-drivers/source/mbed-drivers.a ym/min
ar/source/minar.a ym/core-util/source/core-util.a ym/minar-platform/yotta_dummy_
lib_minar_platform/minar-platform.a ym/minar-platform-mbed/source/minar-platform
-mbed.a ym/mbed-hal-nrf51822-mcu/existing/mbed-hal-nrf51822-mcu.a ym/mbed-hal/so
urce/mbed-hal.a ym/mbed-hal-nordic/yotta_dummy_lib_mbed_hal_nordic/mbed-hal-nord
ic.a ym/mbed-hal-blenano/source/mbed-hal-blenano.a ym/mbed-drivers/source/mbed-d
rivers.a ym/minar/source/minar.a ym/core-util/source/core-util.a ym/minar-platfo
rm/yotta_dummy_lib_minar_platform/minar-platform.a ym/minar-platform-mbed/source
/minar-platform-mbed.a ym/compiler-polyfill/yotta_dummy_lib_compiler_polyfill/co
mpiler-polyfill.a ym/ualloc/source/ualloc.a ym/dlmalloc/source/dlmalloc.a -Wl,-w
rap,_malloc_r -Wl,-wrap,_calloc_r -Wl,-wrap,_realloc_r -Wl,-wrap,_free_r -Wl,-wr
ap,_sbrk_r ym/cmsis-core/source/cmsis-core.a ym/cmsis-core-nordic/yotta_dummy_li
b_cmsis_core_nordic/cmsis-core-nordic.a ym/cmsis-core-nrf51822/source/cmsis-core
-nrf51822.a ym/cmsis-core/source/cmsis-core.a ym/cmsis-core-nordic/yotta_dummy_l
ib_cmsis_core_nordic/cmsis-core-nordic.a ym/cmsis-core-nrf51822/source/cmsis-cor
e-nrf51822.a  -lstdc++ -lsupc++ -lm -lc -lgcc -lstdc++ -lsupc++ -lm -lc -lgcc -W
l,--end-group  --specs=nano.specs -o source/ble-beacon && cmd.exe /C "cd /D "C:/
Users/Nick Guin/ble-examples/BLE_Beacon/build/blenano-gcc/source" && arm-none-ea
bi-objcopy -O ihex ble-beacon ble-beacon.hex && python "C:/Users/Nick Guin/ble-e
xamples/BLE_Beacon/yotta_targets/nordic-nrf51822-16k-gcc/CMake/../scripts/merge_
hex.py" "C:/Users/Nick Guin/ble-examples/BLE_Beacon/yotta_targets/nordic-nrf5182
2-16k-gcc/CMake/../softdevice/s130_nrf51_1.0.0_softdevice.hex" ble-beacon.hex bl
e-beacon-combined.hex && cd /D "C:/Users/Nick Guin/ble-examples/BLE_Beacon/build
/blenano-gcc/source" && C:/yotta/gcc/bin/arm-none-eabi-objcopy.exe -O binary ble
-beacon ble-beacon.bin""
arm-none-eabi-g++.exe: error: Guin/ble-examples/BLE_Beacon/yotta_targets/nordic-
nrf51822-16k-gcc/CMake/../ld/NRF51822.ld: No such file or directory
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

yotta --version
0.8.5

autopulated, 2015-10-26:
I believe this is a problem with string quoting in the target description that you're using.

@rgrover is blenano-gcc yours? / where can I find the source?

rgrover, 2015-10-26:
@autopulated I believe Alessandro or Andres could be behind the  ble-nano-gcc target. It doesn't come from me.

0xc0170, 2015-10-26:
@AlessandroA @andresag01 Please can you publish ble-nano-gcc target here on github? I can't locate it

Looking at the command above, seems like -T is not in quotes:
-TC:/Users/Nick Guin/ble-example
s/BLE_Beacon/yotta_targets/nordic-nrf51822-16k-gcc/CMake/../ld/NRF51822.ld

Taken from toolchain.cmake for nordic devices:
```
set(CMAKE_EXE_LINKER_FLAGS_INIT    "${CMAKE_EXE_LINKER_FLAGS_INIT} -mcpu=cortex-m0 -mthumb -T${CMAKE_CURRENT_LIST_DIR}/../ld/NRF51822.ld")
```
It's missing string quotes. I might send PR fixing or whoever is faster.

0xc0170, 2015-10-27:
PR sent, waiting for review. It should fix your problem @Hydrogencyanide1

closed 2015-10-29 by autopulated