ARMmbed/yotta #529: Require one of homepage, repository or source tarball url when publishing yotta modules.

jrobeson, 2015-10-24:
Currently it's possible to publish yotta modules to the registry without defining the the project home page, source tarball link, or a repository key. I can understand not requiring either of these when making local targets/modules, but I do think that the public yotta registry should enforce having at least one of these keys.

autopulated, 2015-10-26:
I noticed that npm doesn't do this, so my default "if in doubt, copy npm" strategy was not to enforce this.

jrobeson, 2015-10-27:
I wouldn't hold npm as a paragon of good package management system design. :)

In any case though, the lack of these these particular items has made it impossible to find the source code of 3 modules so far.

jrobeson, 2015-10-27:
here's one example right here of someone not being able to find the source and it not being mentioned in the the module.json: https://github.com/ARMmbed/yotta/issues/528#issuecomment-151119608

0xc0170, 2015-10-27:
I would like to see a reference to where the code lives to be able to contribute.

mlnx, 2015-10-27:
+1
There should be a URL to the source code repository (ideally github)

autopulated, 2015-10-27:
All true ;) – but using yotta is not a substitute for people doing the right things.

yotta can't make a module author use github or accept pull requests any more than it could magically make badly written software re-usable.

I think what I'm trying to say is: we need other ways of encouraging people to do the right things, too: relying on restrictions in tools isn't enough unless people understand why those restrictions are there.

That said, I'm not opposed to adding this as gentle encouragement... the place to add it is in the [schema files](https://github.com/ARMmbed/yotta/tree/master/yotta/lib/schema).

jrobeson, 2015-10-27:
I'm not as concerned with what people do with their own private modules or stuff they host outside of the registry btw. I just want the registry to be a useful source of information on its own right.

jrobeson, 2015-12-15:
another instance of the problem. I can find blenano-gcc in the registry, and even download it, but i have zero idea who wrote it or who to contact to add support for the 32kb ram version 1.5

```json
{
  "name": "blenano-gcc",
  "version": "0.0.2",
  "inherits": {
    "nordic-nrf51822-16k-gcc": "*"
  },
  "description": "Unofficial mbed build target for the RedBearLab BLE Nano board",
  "licenses": [
    {
      "url": "https://spdx.org/licenses/Apache-2.0",
      "type": "Apache-2.0"
    }
  ],
  "keywords": [
    "mbed-official",
    "gcc"
  ],
  "similarTo": [
    "rblab",
    "blenano"
  ],
  "config": {}
}
```

jrobeson, 2015-12-15:
oh hah, that was the example I already used!  for blenano. I'm going to leave it up here to show a specific example of the problem though.