ARMmbed/yotta #531: build for multiple target devices at once

jrobeson, 2015-10-25:
Maybe it's just not clear from the documentation, but I can't tell how you can build for multiple target devices at one time.

BlackstoneEngineering, 2015-10-25:
Currently yotta just builds for one target at a time. You could easily script this by running multip retargets and builds though.

Something like 
```
yotta target T1
yotta build
yotta target T2
yotta build
```

jrobeson, 2015-10-25:
yeah, that is what i'm doing right now, except as:

```bash
$ yotta -t T1 build
$ yotta -t T2 build
```

autopulated, 2015-10-26:
yotta should work just fine if use your shell to run the commands in parallel:

```sh
yt -t T1 build &
yt -t T2 build
```

I'm not sure this is really something that should be internal to yotta though, your shell already has a well-built UI for managing multiple concurrent tasks.