ARMmbed/yotta #532: yotta upload or flash command?

jrobeson, 2015-10-25:
I'm not sure if this is the domain of yotta or not. Feel close to close if it isn't

BlackstoneEngineering, 2015-10-25:
try `yotta debug`

jrobeson, 2015-10-25:
I have my own command to flash the devices. It uses udisksctl to find the devices and then simply copies it. I just wanted to know if a direct flash command will be added to yotta or not.

BlackstoneEngineering, 2015-10-25:
No, there are other tools that do this, such as exporting to external toolchains with [valinor](https://github.com/ARMmbed/valinor), or using the [mbed-greentea](https://github.com/ARMmbed/greentea) test bench integrator

jrobeson, 2015-10-25:
external toolchains with valinor? Isn't that what I have already when I"m using yotta?

0xc0170, 2015-10-26:
@jrobeson You are talking about mbed devices which have an interface chip on board with virtual drive, what about a device outside of mbed ? 

Have you tried yotta debug? Do you find it limiting or see an improvements? Share with us

There are many diff ways to flash a device - some IDE does not have that support to flash a device via a command line, or have some limitations (IAR I am pointing to you :-)) or require a configuration file (openOCD) or just work outside of the box but for limited number of devices (pyOCD) or which works only with jlink but for almost any device (jlinkgdb).

jrobeson, 2015-10-26:
I'd assume initial support for mbed boards (where the usb drive shows up), and then jlink.

autopulated, 2015-10-26:
I think there is a place for `yotta run`, which mbed / embedded targets could interpret as being "upload and run". `yotta test` works the same way, for example. I'm closing this as a duplicate of #109

closed 2015-10-26 by autopulated