ARMmbed/yotta #533: Publishing Executables to the registry

BlackstoneEngineering, 2015-10-25:
There should be a warning message when a user tries to publish executables to the yotta registry. Something like
```shell
$ yt publish
Publishing v0.0.1 of My Module...
WARNING: This module is an executables, it is generally a bad idea to publish executables to the
 registry as they are not re-usable. Are you sure you with to publish this executable to the 
 registry? (Y/N)
```

BlackstoneEngineering, 2015-10-25:
Something, anything, to somehow discourage this behaviour. I would prefer outright rejection of executables from the registry, but it is what it is.

autopulated, 2015-10-26:
Yeah, there should be: applications can't be depended on by other modules, so the utility of publishing them is limited.

PR #544 (2015-10-30): warn when publishing applications; make applications private by default
fixes #533
closed 2015-10-30 by autopulated