ARMmbed/yotta #534: Add yotta size command

niklas-arm, 2015-10-26:
`yt size` would display the static Flash and RAM usage of the executable very similar to how `arm-none-eabi-size` commands outputs this information.

This would give you an idea how much RAM is used and what performance can be expected.

autopulated, 2015-10-26:
I'm cautious about adding a yotta subcommand for this, but it is something that targets could possibly provide a script to do (the way you need to do this is target-specific anyway).

Note that statically determinable flash/ram use is only part of the story: your dynamic RAM use can only really be determined by profiling.

niklas-arm, 2015-10-26:
Yes, a script in the target was also how I imagined this being implemented, to keep the yotta code target-independent.

[profiling]: yes, but any information is better than no information.
If you know your RAM is 90% full with static data, you'll have a much easier time understanding why your `malloc` is failing (and then asserting).

autopulated, 2015-10-26:
true ;)