ARMmbed/yotta #535: [master branch] DIM character in yotta search command with --plain mode under Ubuntu

PrzemekWirkus, 2015-10-26:
## Descriptiom
While running:
```
$ yotta --plain search -k mbed-target:k64f target
```
```
frdm-k64f-gcc 0.2.0:
    Official mbed build target for the mbed frdm-k64f development board.
    mbed-target:k64f, mbed-official, k64f, frdm-k64f, gcc
frdm-k64f-armcc 0.1.4:
    Official mbed build target for the mbed frdm-k64f development board, using the armcc toolchain.
    mbed-target:k64f, mbed-official, k64f, frdm-k64f, armcc

additional results from https://yotta-private.herokuapp.com:
```
I've parsed [DIM](http://plainbox.readthedocs.org/en/latest/ref/plainbox.impl.color.html#plainbox.impl.color.ansi_on.s.DIM) control character in line with target name:
```
frdm-k64f-gcc 0.2.0:
```
Parsed target name:
```python
['frdm-k64f-gcc\x1b[2m']
```
## Comment
This causes Greentea under Ubuntu Linux to not correctly parse target name.

autopulated, 2015-10-26:
Thanks for reporting.

autopulated, 2015-10-30:
I'm unable to reproduce this. If I pipe the output into, for example `sed -n 'l'` which will display the control characters, I don't see any when --plain is used.

closed 2015-10-30 by autopulated