ARMmbed/yotta #538: Safeguards for `yotta publish`

bogdanm, 2015-10-28:
Creating a new version with `yt version` is a multi-step process:

1. create the version itself
2. push the resulting commit
3. push the new tag
4. publish the new version

In practice, people often forget to do one of more of the steps above, most often 2 or 3. It would be good if `yt version` and/or `yt publish` could warn (or remind) people what they need to do.

autopulated, 2015-10-28:
You don't need to push the tag anywhere for things to work correctly – that's only a convenience for you. (Actually the same for pushing the commit).

It isn't yotta's job to manage your VCS workflow!

bogdanm, 2015-10-28:
> It isn't yotta's job to manage your VCS workflow!

While that's tehcnically true, I still feel yotta can help a bit here.

autopulated, 2015-10-28:
```
mkdir -p .git/hooks && echo '#/bin/sh
echo "always remember to push!"
' > .git/hooks/post-commit && chmod +x .git/hooks/post-commit
```

;)

bogdanm, 2015-10-28:
:smile: