ARMmbed/yotta #539: Require a way to log in to github

thegecko, 2015-10-28:
When I use a github dependency in yotta and am behind a proxy, github can reach anonymous download limits and refuse to serve requests:

```
github.GithubException.GithubException: 403 {u'documentation_url': u'https://developer.github.com/v3/#rate-limiting', u'message': u"API rate limit exceeded for 81.134.128.130.
```

Can yotta be made to allow github logins?

autopulated, 2015-10-28:
Thanks for reporting – when this error gets returned, yotta should prompt you to log in automatically.

markus-becker-tridonic-com, 2015-11-03:
We are hit by this as well. Is there a possibility of a workaround with GitHub's personal access tokens?

autopulated, 2015-11-03:
If you are accessing private repos the issue won't affect you – so the best workaround is to publish to the registry the things that you are pulling from public githubs.

I'm currently investigating the proper fix for this though, stay tuned!

PR #553 (2015-11-03): improve github authentication handling decorator:
( treat rate-limit-exceeded errors as requiring auth, and reduce code duplication)

Fixes #539 

cc @thegecko
closed 2015-11-03 by autopulated