ARMmbed/yotta #541: OSError: [Errno 17] File exists from multiprocessing/synchronize.py

stevenewey, 2015-10-29:
In testing the compile service for the micro:bit I ran a test that performed 51,200 builds in less than 10 minutes. 2 of those builds failed with this error. It may be hard to recreate!

```
\x1b[22m\x1b[32minfo\x1b[39m: \x1b[22m\x1b[39mgenerate for target: bbc-microbit-classic-armcc 0.1.0 at /build/yotta_targets/bbc-microbit-classic-armcc\x1b[0m', 'Traceback (most recent call last):
  File "/usr/local/bin/yotta", line 4, in <module>', 'yotta.main()', 'File "/venv/local/lib/python2.7/site-packages/yotta/main.py", line 229, in main', 'status = args.command(args, following_args)
  File "/venv/local/lib/python2.7/site-packages/yotta/build.py", line 41, in execCommand
    status = installAndBuild(args, following_args)
  File "/venv/local/lib/python2.7/site-packages/yotta/build.py", line 118, in installAndBuild
    build_args=following_args, targets=args.build_targets
  File "/venv/local/lib/python2.7/site-packages/yotta/lib/fsutils_posix.py", line 57, in wrapped_fn
    q = multiprocessing.Queue()
  File "/usr/lib/python2.7/multiprocessing/__init__.py", line 218, in Queue
    return Queue(maxsize)
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 63, in __init__
    self._rlock = Lock()
  File "/usr/lib/python2.7/multiprocessing/synchronize.py", line 147, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1)
  File "/usr/lib/python2.7/multiprocessing/synchronize.py", line 75, in __init__
    sl = self._semlock = _multiprocessing.SemLock(kind, value, maxvalue)
OSError: [Errno 17] File exists
```

autopulated, 2015-10-30:
This is a bug in python: https://bugs.python.org/issue24303 It seems that it is fixed in python >= 3.4.

yotta should support python 3.4, though it isn't as well tested as python 2.7 support.

stevenewey, 2015-10-30:
OK, thanks. Interestingly I've done several hundred thousand compiles this morning and not seen it once. I'll investigate Python 3 for the toolbox at some point.