ARMmbed/yotta #542: Feature request: yotta deprecate <pkg>[@version] <message>

AlessandroA, 2015-10-30:
As a module or target owner I would like to be able to mark a component as obsolete, so that a warning can issued to someone that uses it or installs it from the public registry.

Most of the cases related to modules obsolescence are solved by semver, but the following are not:
- a module has been permanently renamed; of course the package with the old name still stays in the registry, but the owner probably wants to promote the new one
- a module is no longer supported by the developer, and (possibly) an equivalent exists

I think this would be equivalent to `npm deprecate <pkg>[@<version>] <message>`.

Ideally, a new "redirect" property could point to a valid replacement for the module. We could also mimic the behavior of GitHub, which automatically redirects to renamed modules (of course if semvers match and the owner allows it explicitly).

autopulated, 2015-10-30:
You could simulate this behaviour by turning your old module into a shim module which depends on your old module, but has a .cmake file in the empty ./source directory that displays a message about being deprecated.

AlessandroA, 2015-10-30:
Ah, makes sense. You can reject this then, I think the approach you suggested does pretty much the same. Do you think it should be documented? Like a "Deprecating a module" section in the docs. I guess this is something that is going to happen often (we are already changing some module names), would be nice to have it consistent

autopulated, 2015-10-30:
We should definitely write a guide on how to do that sort of thing.

It could live in an advanced tutorial / how-to section of the yotta docs. It might be a good idea to split the current tutorial section into simple/advanced ones anyway.