ARMmbed/yotta #543: Cosmetic: allow `yt install` to run without error on windows

mpg, 2015-10-30:
See https://github.com/ARMmbed/mbedtls/issues/333

Proposed resolutions:
1. Add an option '-D, --no-dependencies' to download only the module without its dependencies
2. Add an option --t, --target' to set the target used for resolving dependencies to download
3. have a default target on windows (might still cause some issue when installing packages that hard-depend on mbed OS and will probably have missing deps with the default windows target)

Note: there is a work-around to this issue: set the default target to a suitable value before running `yt install`.

autopulated, 2015-10-30:
The `-t` option already exists: `yotta -t <targetname> subcommand` can be used with all subcommands.

The best fix would be to have a default windows target though.

mpg, 2015-10-30:
Oh, that's great, I didn't realise that (I'll use that in my build tests, currently I was using `yt target && yt build` which is not very satisfying). I'm now wondering if it could be a good idea to repeat global options in the help message for each subcommand (`-h` seems to be repeated).

autopulated, 2015-10-30:
@mpg yes, it'd also be nice for them to be usable after the subcommand name (at the moment they have to go before the subcommand): this would make them show up in help too.