ARMmbed/yotta #551: yotta update does not complete update process

tabarr, 2015-10-30:
I run "yt ls" and see version mismatches in my project (due to the 1.0.0 reversioning). I then run "yt update" and see updates being downloaded. But when I run "yt ls" again, I still see version mismatches. The only way I can get a new version to update properly is to delete the module directory. The "yt ls" shows the module being missing and "yt update" successfully downloads and installs the new module.

(workspace) c:\mbedos\projects\mbed-client-example-6lowpan>yt update
info: get versions for frdm-k64f-gcc
info: get versions for mbed-gcc
info: get versions for mbed-client
info: get versions for mbed-mesh-api
info: get versions for mbed-client-c
info: download mbed-client-c@1.1.1 from the public module registry
info: get versions for mbed-client-mbed-os
info: get versions for mbed-client-mbedtls
info: get versions for nanostack-libservice
info: get versions for mbed-drivers
info: get versions for sockets
info: download sockets@1.0.2 from the public module registry
info: get versions for mbed-hal
info: get versions for cmsis-core
info: get versions for ualloc
info: get versions for minar
info: get versions for core-util
info: get versions for compiler-polyfill
info: get versions for mbed-hal-freescale
info: get versions for mbed-hal-ksdk-mcu
info: get versions for uvisor-lib
info: download uvisor-lib@1.0.4 from the public module registry
info: get versions for mbed-hal-k64f
info: get versions for cmsis-core-freescale
info: get versions for cmsis-core-k64f
info: get versions for mbed-hal-frdm-k64f
info: get versions for dlmalloc
info: get versions for minar-platform
info: get versions for minar-platform-mbed
info: get versions for sal
info: download sal@1.0.2 from the public module registry
info: get versions for mbedtls
info: get versions for sal-iface-6lowpan
info: get versions for sal-stack-nanostack
info: get versions for sal-stack-nanostack-eventloop
info: download sal-stack-nanostack-eventloop@1.0.3 from the public module registry
info: get versions for nanostack-randlib
info: get versions for mbed-6lowpan-eventloop-adaptor
info: download mbed-6lowpan-eventloop-adaptor@1.0.2 from the public module registry(workspace) 

c:\mbedos\projects\mbed-client-example-6lowpan>yt ls
mbed-client-example-6lowpan 0.0.2
|_ mbed-client 1.1.15
| |_ mbed-client-c 1.0.6 >=1.1.0,<2.0.0 yotta_modules\mbed-client-c
| | \_ nanostack-libservice 3.0.7 yotta_modules\nanostack-libservice
| |_ mbed-client-mbed-os 1.0.11 yotta_modules\mbed-client-mbed-os
| | |_ mbed-drivers 0.11.1 yotta_modules\mbed-drivers
| | | |_ mbed-hal 1.0.3 yotta_modules\mbed-hal
| | | | \_ mbed-hal-freescale 1.0.0 yotta_modules\mbed-hal-freescale
| | | |   \_ mbed-hal-ksdk-mcu 1.0.3 yotta_modules\mbed-hal-ksdk-mcu
| | | |     |_ uvisor-lib 0.9.0 >=1.0.0,<2.0.0 yotta_modules\uvisor-lib
| | | |     \_ mbed-hal-k64f 1.0.1 yotta_modules\mbed-hal-k64f
| | | |       \_ mbed-hal-frdm-k64f 1.0.0 yotta_modules\mbed-hal-frdm-k64f
| | | |_ cmsis-core 1.0.1 yotta_modules\cmsis-core
| | | | \_ cmsis-core-freescale 1.0.0 yotta_modules\cmsis-core-freescale
| | | |   \_ cmsis-core-k64f 1.0.0 yotta_modules\cmsis-core-k64f
| | | |_ ualloc 1.0.2 yotta_modules\ualloc
| | | | \_ dlmalloc 1.0.0 yotta_modules\dlmalloc
| | | |_ minar 1.0.1 yotta_modules\minar
| | | | \_ minar-platform 1.0.0 yotta_modules\minar-platform
| | | |   \_ minar-platform-mbed 1.0.0 yotta_modules\minar-platform-mbed
| | | |_ core-util 1.0.1 yotta_modules\core-util
| | | \_ compiler-polyfill 1.1.1 yotta_modules\compiler-polyfill
| | \_ sockets 0.6.0 >=1.0.0,<2.0.0 yotta_modules\sockets
| |   \_ sal 0.4.1 yotta_modules\sal
| |     \_ sal-stack-lwip 1.0.1 >=0.4.0,<0.5.0 yotta_modules\sal-stack-lwip -> c:\mbedos\dev-modules\sal-stack-lwip
| |       |_ sal-driver-lwip-k64f-eth 0.2.0 >=1.0.0,<2.0.0 yotta_modules\sal-driver-lwip-k64f-eth
[... 220 lines cut ...]
| | | |   \_ cmsis-core-k64f 1.0.0 yotta_modules\cmsis-core-k64f
| | | |_ ualloc 1.0.2 yotta_modules\ualloc
| | | | \_ dlmalloc 1.0.0 yotta_modules\dlmalloc
| | | |_ minar 1.0.1 yotta_modules\minar
| | | | \_ minar-platform 1.0.0 yotta_modules\minar-platform
| | | |   \_ minar-platform-mbed 1.0.0 yotta_modules\minar-platform-mbed
| | | |_ core-util 1.0.1 yotta_modules\core-util
| | | \_ compiler-polyfill 1.1.1 yotta_modules\compiler-polyfill
| | \_ sockets 1.0.2 yotta_modules\sockets
| |   \_ sal 1.0.2 yotta_modules\sal
| |     \_ sal-stack-lwip 1.0.1 yotta_modules\sal-stack-lwip -> c:\mbedos\dev-modules\sal-stack-lwip
| |       |_ sal-driver-lwip-k64f-eth ^1.0.0 missing
| |       \_ sal-iface-eth 1.0.0 yotta_modules\sal-iface-eth -> c:\mbedos\dev-modules\sal-iface-eth
| \_ mbed-client-mbedtls 1.0.6 yotta_modules\mbed-client-mbedtls
|   \_ mbedtls 2.2.0-rc.1 yotta_modules\mbedtls
|_ atmel-rf-driver 1.0.1 >=0.0.17,<0.1.0 -> C:\mbedos\dev-modules\atmel-rf-driver
| \_ sal-stack-nanostack 3.0.5 yotta_modules\sal-stack-nanostack
|   |_ sal-stack-nanostack-eventloop 1.0.3 yotta_modules\sal-stack-nanostack-eventloop
|   | \_ mbed-6lowpan-eventloop-adaptor 1.0.2 yotta_modules\mbed-6lowpan-eventloop-adaptor
|   \_ nanostack-randlib 1.0.0 yotta_modules\nanostack-randlib
\_ mbed-mesh-api 0.3.6
  \_ sal-iface-6lowpan 0.2.15 yotta_modules\sal-iface-6lowpan

sal-iface-lwip-k64-eth still missing but is reported in another issue
atmel-rf-driver mismatch I assume needs to be corrected as part of the reversioning to 1.0.0 process.

autopulated, 2015-10-30:
`yotta update` will only update things so that they still match the version specifications in the modules that depend on them (http://yottadocs.mbed.com/reference/module.html#dependencies)

you can use `yotta ls -a` to see which modules are imposing version requirements on other modules, it's hard to see what's causing your problems above without `yt ls -a` output, but probably some of your dependencies are holding other ones back.

Also note that you have atmel-rf-driver linked: yotta won't modify your linked module, so to update that one you will need to do a `git pull` to update your linked copy.

tabarr, 2015-10-30:
Examples from above:

Called "yt update" and see this action taken
info: download mbed-client-c@1.1.1 from the public module registry

But "yt ls" result after is :
| |_ mbed-client-c 1.0.6 >=1.1.0,<2.0.0 yotta_modules\mbed-client-c

So supposedly mbed-client-c version was updated to 1.1.1 by "yt update" but "yt ls" still reports 1.0.6

I then delete mbed-client-c from yotta-modules directory and run "yt ls"

| |_ mbed-client-c ^1.1.0 missing

Then run "yt update" and "yt ls" again and see this

| |_ mbed-client-c 1.1.1 yotta_modules\mbed-client-c

Basically, what I am seeing is that some modules are not being update properly unless I remove them completely.

autopulated, 2015-11-02:
Thanks – it looks like you have some linked modules which depend on the module in question, so I think this is related to #318