ARMmbed/yotta #556: Passing CMake options to dependent "existing" module CMake

lws-team, 2015-11-04:
I have a small executable project that has a dependent module that is actually a CMake project of its own.

Yotta does well understanding that it is an "existing" CMake project and building it.

But I need to configure the CMake options of the dependent module from the executable project, I could not find a way to pass -DFEATURE=X type CMake defines into the dependent module.

Does this exist or must the dependent module CMakeLists.txt be hacked... it's a bit yucky when the preexisting library also builds for other platforms using CMake options.

autopulated, 2015-11-04:
You could put your module inside a directory with its own CMakeLists.txt file, which sets the options and then calls add_subdirectory on the existing module:

```
modulename
   |_ CMakeLists.txt        # set(...), add_subdirectory(existingmodule)
   \_ existingmodule
         |_CMakeLists.txt
        ...
```

Is it a module you would wrap up for other people to use?

When modifying the existing CMakeLists.txt you could guard the defaults you want to set when building for yotta in tests for the version definition for your module that yotta will define:

```
if(DEFINED YOTTA_<MODULENAME>_VERSION_STRING)
   set(FEATURE=X)
endif()
```

lws-team, 2015-11-04:
Yes it's an existing library targeting other (posixy) platforms already, I'm basically 'adding mbed3 support' to the existing project.  https://libwebsockets.org

Your second method sounds good to me, I'll give it a try thanks.

closed 2015-11-04 by lws-team

autopulated, 2015-11-04:
OK cool – don't hesitate to ask if you have any more qeries!