ARMmbed/yotta #559: AuthenticationFailed on yotta install

markus-becker-tridonic-com, 2015-11-04:
When doing ```yotta install``` on a Linuc machine with yotta 0.9.2 I do get a AuthenticationFailed.

```
info: get versions for x86-linux-native
info: Starting new HTTPS connection (1): registry.yottabuild.org
info: download x86-linux-native@1.0.0 from the public module registry
info: Starting new HTTPS connection (1): registry.yottabuild.org
info: Starting new HTTPS connection (1): yotta.blob.core.windows.net
error: server returned status 403: ﻿<?xml version="1.0" encoding="utf-8"?><Error><Code>AuthenticationFailed</Code><Message>Server failed to authenticate the request. Make sure the value of Authorization header is formed correctly including the signature.
RequestId:1b2101e4-0001-00ba-0f3b-176731000000
Time:2015-11-04T19:59:46.1752168Z</Message><AuthenticationErrorDetail>Request date header not specified</AuthenticationErrorDetail></Error>
Fatal Exception, yotta=0.9.2
Traceback (most recent call last):
  File "/usr/local/bin/yotta", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 240, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 59, in execCommand
    return installDeps(args, c)
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 86, in installDeps
    target, errors = current_component.satisfyTarget(args.target)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 580, in satisfyTarget
    update_installed = update_installed
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/target.py", line 95, in getDerivedTarget
    type = 'target'
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 352, in satisfyVersion
    return satisfyVersionByInstalling(name, version_required, working_directory, type=type)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 297, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 308, in _satisfyVersionByInstallingVersion
    version.unpackInto(working_directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 443, in unpackInto
    _getTarball(self.url, directory, self.sha256)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 102, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 185, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 168, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 149, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 289, in _getTarball
    response.raise_for_status()
  File "/usr/lib/python2.7/dist-packages/requests/models.py", line 773, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 403 Client Error: Server failed to authenticate the request. Make sure the value of Authorization header is formed correctly including the signature.
```

autopulated, 2015-11-05:
Thanks for reporting this – are you still experiencing the problem if you try again now? I've been unable to reproduce it locally.

markus-becker-tridonic-com, 2015-11-05:
Have not seen this again today. Will report again, if I see it again.

closed 2015-11-05 by markus-becker-tridonic-com